riscvarchive/riscv-v-spec #545: Zvamo precludes existence of .B/.Q2/.Q4/.Q8 AMOs

jrtc27, 2020-08-01:
Currently, Zvamo re-uses some of the spare AMO widths for its own purposes, namely .B, .Q2, .Q4 and .Q8. I believe this is a mistake that will seriously limit some implementations.

Firstly, I regard not having .B/.H AMOs, or at least LR/SC, a deficiency in the base ISA. It _can_ be worked around with masking and shifting, but microarchitecturally LR/SC are basically free to implement, as reservations will likely be either at cache line granularity or just a register with an address, and the byte range selection logic is already there for LB and LH. I therefore think a Zabh extension would make a huge deal of sense to add (especially given _Atomic(_Bool) is quite a common construct), just like how the some parts of the bitmanip extensions add opcodes for combinations that are already supported by the ALU.

There is also significant belief that 2*XLEN AMOs are important to have for lock-free data structures, and most successful performant ISAs have them. That would mean LR.D on RV32, LR.Q on RV64, and LR.Q2 on RV128.

Moreover, CHERI (http://cheri-cpu.org), which currently has MIPS and RISC-V implementations from Cambridge, and an upcoming experimental prototype Armv8 taped-out implementation (which many are betting on as being the future of hardware security in the industry) from Arm themselves in collaboration with us (https://developer.arm.com/architectures/cpu-architecture/a-profile/morello), introduces additional issues. On CHERI-RISC-V, we fundamentally need .B and .H LR/SC (we can still do other AMOs by open-coding based on the _non_-masked correct width LR/SC), as when capability mode is turned on, rs1 is a capability with fine-grained bounds, i.e. only 1 byte for a char, and 2 bytes for a short, so using .W and masking would trap with an out-of-bounds exception. Our capabilities themselves are 2*XLEN in size, and we need AMOs for them too, so we need to use LR.D/LR.Q already on RV32/RV64, and LR.Q2 if RV128 ever ends up being a meaningful thing. Plus, if we were to want to support atomic pair operations as discussed above, we would need to use LR.Q/LR.Q2/LR.Q4, leaving _just_ LR.Q8 unused. Note that Armv8 includes atomic pair operations, which would be Q for integers and Q2 for capabilities in RISC-V parlance on RV64, and we believe that any mature production-ready CHERI implementation should have them for capabilities, just as any mature production-ready non-CHERI implementation should have integer pair atomics for traditional pointers (i.e. where a pointer is just an XLEN address). But since they don't currently exist for RISC-V, we don't currently add them for CHERI-RISC-V.

For all of these reasons I would very much suggest you choose a different encoding space for the AMOs. Whilst it may be the most convenient option right now, I believe it significantly limits future options and will end up causing problems when discovering that the stolen AMO encodings are needed for their original purposes.

jrtc27, 2020-11-13:
FWIW my suggestion would be to define new VAMO* funct5's within OP-AMO, as there's plenty of space in there. Unfortunately there isn't a "nice" way to encode them by say setting a single bit to turn AMO* into VAMO*, but you could get close with something like the following:

|     | 00      | 01       | 10      | 11       |
|-----|---------|----------|---------|----------|
| 000 | AMOADD  | AMOSWAP  | LR      | SC       |
| 001 | AMOXOR  | VAMOXOR  | VAMOADD | VAMOSWAP |
| 010 | AMOOR   | VAMOOR   |         |          |
| 011 | AMOAND  | VAMOAND  |         |          |
| 100 | AMOMIN  | VAMOMIN  |         |          |
| 101 | AMOMAX  | VAMOMAX  |         |          |
| 110 | AMOMINU | VAMOMINU |         |          |
| 111 | AMOMAXU | VAMOMAXU |         |          |

VAMOADD and VAMOSWAP are the awkward ones to deal with due to the encodings of LR and SC.

Then you can just use the width codes corresponding to the actual element width you want, which is also nicer than the current draft encodings, and there's room to grow more widths too for, say, E128 through E1024, as available for the normal loads/stores, if that ever becomes a useful thing (E128 at least might, especially on RV128).

aswaterman, 2020-11-13:
A lower-friction alternative at this point would be, if we do later decide to define byte/half-word AMOs, to do so in the MISC-MEM encoding space.

jrtc27, 2020-11-13:
From your perspective, sure. For us, that would be a huge nuisance, since we _require_ the byte/half-word AMOs and have 7 implementations of CHERI-RISC-V (QEMU, Spike and Sail simulators/emulators, and Piccolo, Flute, Toooba and Ibex soft-cores), plus the 2*XLEN AMOs, and used what we thought the logical encodings for those would be. Given that my proposal is cleaner than the current draft, avoids forcing us to re-encode a chunk of our instructions, can be folded in with any other 0.9->1.0 breaking changes, and there are unlikely to be many, if any, significant consumers of the current Zvamo encodings (unlike us, where we already have companies (including Microsoft and Google) and other universities looking at and performing research atop CHERI, many of whom are using CHERI-RISC-V rather than Morello due to its open-source nature, and so any large encoding changes like that cause significant friction). I'm of course biased, but even despite that I'd still believe using different funct5's is the "right" way to do it, though ultimately whatever the RISC-V Foundation says is what we have to deal with whether we like it or not.

Rolfy360, 2020-11-13:
Use the encoding method suggested by jrtc27,
there are 32 instructions in each width, from e8 to e1024

[image]

kasanovic, 2021-03-23:
There has been considerable recent discussion around this issue.   I think the proposal emerging is to rework the vector AMO encoding to avoid stomping on the natural place to add the subword atomics.  It will take a little while to figure out good recoding of vector AMO, possibly trying to incorporate #457.  To continue to make progress, we can drop vector AMO as a required part of application processor vector extension in v1.0.

David-Horner, 2021-03-26:
I believe a transient vsetvl[i] #456 lock setting would allow  an sequence of vector instructions.
something like:

vlock 3,mask_reg # number of following vector ops and a mask of "successfully locked'
vl
vop
vs under mask_reg #  attempt write back using OK so far mask, and reset the mask element if atomic write unsuccessful

I agree that the best cannot be worked out before v1.0.

commit a8098bd1 (2021-06-07): Removed Zvamo from standard extension.
Will be added as a separate vector extension at a later date but with a new encoding.
Closes #545.
closed 2021-06-07 by kasanovic