riscvarchive/riscv-v-spec #546: Need sub-extension for baseline vector ISA

kito-cheng, 2020-08-03:
Currently Zvamo and Zvlsseg are included in the base "V" extension, which mean there is no way to disable those two sub-extension by any ISA string combination.

So I think we need a sub-extension name for `V - (Zvamo + Zvlsseg)`, maybe `zvbase`.

jim-wilson, 2020-08-03:
What the riscv-v-spec text actually says is "This set of instructions is included in the base "V" extension used for the Unix profile.".  This implies that Zvamo is not part of the base V extension, but is part of the Linux platform specification.  However, internal SiFive users are expecting the toolchain to always enable Zvamo when the v extension is enabled.  This is confusing.  The toolchain folks are getting mixed messages about whether zvamo is part of the base V extension or not and need clarification.

If it is part of the base V spec, then we need a zvbase like Kito suggested.

Likewise for zvlsseg.

jim-wilson, 2020-08-03:
OK, I see the zvamo/Unix profile stuff is #547 so we can discuss that there.

kasanovic, 2021-06-07:
Zvamo removed from v1.0 draft, and this issue is now tracked on #547 (which also might be closed soon), so closing this issue.

closed 2021-06-07 by kasanovic