riscvarchive/riscv-v-spec #547: Clarify note for Zvamo and Zvlsseg

kito-cheng, 2020-08-03:
There is note for Zvamo and Zvlsseg as bellow:
```Note: This set of instructions is included in the base "V" extension used for the Unix profile.```

However that's little bit confusion about meaning of `for the Unix profile.`, does it mean V included Zvamo and Zvlsseg for `-linux-gnu` toolchain, but not included for bare-metal toolchain?

https://github.com/riscv/riscv-binutils-gdb/pull/224#issuecomment-666166925

Hsiangkai, 2020-08-07:
Do vector sub-extensions imply 'V' extension? If Zvamo implies 'V' and 'V' implies Zvamo, is there any need to create a sub-extension for Zvamo? On the other hand, if Zvamo does not imply 'V', is it useful to implement Zvamo without 'V' extension?

jim-wilson, 2020-08-07:
See #546 which proposes adding a Zvbase or Zvb or similar.  See also the bitmanip extension which defines a Zbb which is the bitmanip base, and b will be equivalent to Zbb plus possibly a few other Zb* extensions, the exact list hasn't been defined yet.  It would be useful for the toolchain if the vector extension worked the same way, e.g. v = Zvbase + Zvamo + Zvlsseg.  Then someone who didn't implement Zvamo or Zvlsseg can use Zvbase to describe what the hardware supports.  In this case, Zvamo implies Zvbase but not Zvlsseg.

This question is specifically about what "for the Unix profile" means though.

Hsiangkai, 2020-08-10:
Thanks, @jim-wilson. I will follow up #546.

kasanovic, 2021-06-07:
Zvamo is no longer in standard vector extensions.  Will be added later as an extension.

kasanovic, 2021-06-07:
Zvlsseg is part of standard extension.

I think this issue is closable now.

kito-cheng, 2021-06-07:
Thanks for update!

closed 2021-06-07 by kito-cheng