riscvarchive/riscv-v-spec #550: Names for embedded  vector extension

kasanovic, 2020-08-07:
Single-letter “V” is reserved for the app processor ISA extension, which is quite comprehensive but is meant to be stable for app software.

For embedded platforms that want a lot more configurability, we have “Zve” (can read as “embedded vector”) names giving largest supported element width:
Zveb  INT8
Zveh  INT8, INT16
Zvew INT8, INT16, INT32
Zved  INT8, INT16, INT32, INT64

Add a trailing f for IEEE floating-point support
Zvewf INT8, INT16, INT32, FP16, FP32
Zvedf INT8, INT16, INT32, INT64, FP16, FP32, FP64

Some folks want to exclude narrower element widths in which case we use “Zveo..” (o for “only”):
Zveoh 16b
Zveohw 16b,32b

There might be other instructions omitted from base embedded vector extensions (E.g, AMOs).

David-Horner, 2020-08-07:
I believe this smithing work can be done on the list.
Both the wordsmithing  and the hammering out of the scope and details.

So, to start off:

1) I agree that the Zv**** strings should be relevant to profiles/domains-of-application
    I would prefer a bounded profile/domain name rather than a single positional character.

2) I don't believe embedded is distinct in this matter.
 I believe many characteristics will be shared amount profiles/domains and as a result we will end up with considerable redundancies of the form V<profile/domain>{<miriad-of-common-options>}n0{o<miriad-of-less-common-exclusions>}n1
with n0 and n1 repeat factors starting at zero and 1 respectively.

Even with that, I wonder how we can avoid aliasing the many options.

3) We have not discussed , nor raised the issue of limited support for EEW of widened operations. 
That is, ELEN is limited in its use to the target of a widening operation or source of a narrowing.
4) Or the related issue of limited support for such an ELEN in loads and stores.

5) which leads us to the election within Embedded (and others) to limit the load/store options.
  Perhaps to only the unit-stride. Or excluding indexed or segmented. 

These are the ones off the top of my head.
Hopefully to be read before the meeting.

kasanovic, 2020-08-07:
Related to issue #360

kito-cheng, 2020-08-10:
I would suggest the max ELEN should encoded in arch string even for non-embedded platforms, otherwise it's hard to know the minimal requirement for a library/executable, the only way is run and see got exception or not.

kasanovic, 2020-10-09:
Concrete proposal for first set of Zv subsets (others can be defined, now or later):

Zve32x. (ELEN=32, integer+fixed only)
Zve32f   (ELEN=32, int+fixed+FP16+FP32)
Zve64x  (ELEN=64, integer+fixed only)
Zve64f (ELEN=64, integer+fixed up to 64b, FP16+FP32)
Zve64d  (ELEN=64, int+fixed+FP16+FP32+FP64)

These would support defining e.g., RV64IMAC_Zve32x (64b embedded core with integer and fixed point only and vector ops on up to 32b values).

A question is whether 64b integer multiply is included in Zve64f.

Probably exclude Zvamo from these, otherwise every relevant instruction type  in "V".

David-Horner, 2020-10-21:
Is this something that can be done during ratification?

As long as v1.0 has defined the components, how they are sliced and diced appears to be more of a platform concern and not on the critical path to v1.0 release for ratification.

I see this as a larger concern than just for V-extension. It is applicable to all the TG groups.
Such subdivision and target platform allocation does not appear to me as mandatory for  release.
Certainly a review as in B has been helpful, but my take:
the detained refinement, indeed the Z* effort in its entirety, should be moved to another, perhaps more light weight,  ratification process.
They are not ISA standards per se. [perhaps those were fighting words].

kasanovic, 2020-12-18:
Consensus was to drop FP16 from base V and Zve*, as can add either FP16 or later BF16 (or both) as extensions.

Zve32x. (ELEN=32, integer+fixed only)
Zve32f (ELEN=32, int+fixed+FP32)
Zve64x (ELEN=64, integer+fixed only)
Zve64f (ELEN=64, integer+fixed up to 64b, FP32)
Zve64d (ELEN=64, int+fixed+FP32+FP64)

kito-cheng, 2020-12-19:
Hi @kasanovic :
Do you mind clarify the meaning of `drop FP16 from base V and Zve*` ? Does it mean vector FP16 operation can't use under `rv64gv_zfh`, still need extra extension like `zvfp16` extension to enable that?

And more follow up question:
- does it mean zve32x don't have floating point vector operation support even `rv32imafd_zve32x`? 
- does it mean zve32f will have floating point  vector operation support even `rv32i_zve32f`? 
- Does it allow XLEN <= VLEN, e.g. `rv64i_zve32x` or `rv64i_zve32f` is valid configure or not?

commit 808a6f83 (2021-06-07): Added description of vector extensions intended for version 1.0 ratification.
Closes #550.
closed 2021-06-07 by kasanovic