riscvarchive/riscv-v-spec #553: mstatus.vs reset value and security

sorear, 2020-08-08:
For security reasons (preventing storage channels) when running vector-naive operating systems, if supervisor mode is implemented all user-mode vector operations need to trap until the supervisor enables the vector extension, in other words:

* VS must not be automatically set to Dirty if it were previously Off (a 1-bit implementation where all non-Off values immediately become Dirty is still possible)
* The initial state of VS on entry to supervisor code (ideally, this would also be the reset value) must be Off

While not strictly necessary, I would prefer that this apply to all implementations, since the vector extension does not otherwise interact with the existence of supervisor mode and it would be very easy to miss this while adding supervisor mode support to an existing vector-capable core.

kasanovic, 2021-03-19:
I believe this is instance of more general problem.  The SBI implementation should not enable any optional extensions unless explicitly requested by the supervisor code.  The also implies hardware has to be provide means to disable any optional extensions.

kasanovic, 2021-06-07:
This issue is now proposed to be handled in a more general way with the staten proposal from @jhauser-us on the tech-privileged mailing list, so closing issue here.

closed 2021-06-07 by kasanovic