riscvarchive/riscv-v-spec #555: Does Zvamo imply 'A' extension?

Hsiangkai, 2020-08-10:
If Zvamo implies A extension, rv32i_zvamo is a valid combination. Zvamo will turn on ‘A’ extension implicitly.
If Zvamo does not imply A extension, users need to specify rv32ia_zvamo to support Zvamo.

There is a description about the relationship. “If vector AMO instructions are supported, then the scalar Zaamo instructions (atomic operations from the standard A extension) must be present.” What does it mean “must be present”?

Nelson1225, 2020-08-13:
The current GNU assembler choose the second one - Zvamo does not imply A extension.  But is doesn't mean the current behavior of assembler is right.  I'm OK with both (imply or non-imply).

FYI, @kito-cheng reminds me that the ISA spec had the imply list for all extensions [1].  It would be great if we can also have the similar imply list for vector spec.

[1] https://github.com/riscv/riscv-isa-manual/blob/draft-20200812-1eaebcc/src/naming.tex#L157

Thanks
Nelson

kasanovic, 2020-10-23:
Zvaamo should imply Zaamo rather than entire A, which also includes Zalrsc.

kasanovic, 2021-06-07:
Zvamo is removed from the initial standard vector extensions.  These will be added at a later date as a standard extension but need new instruction encodings.