riscvarchive/riscv-v-spec #559: Have to do overlapping and alignment checks in VRGATHEREI16.vv

Rolfy360, 2020-08-14:
case1: VLEN=128, VSEW=8, VMUL=1
vrgatherei16 vd, vs2, vs1
vs1 contains 16 e16  elements(fits in 2 VPRs), so vs1 should be even.

case2: VLEN=128, VSEW=64, VMUL=4
vrgatherei16 v8, v2, vs1
vs1 contains 8 e16 elements(fits in 1 VPR) and can not be v8, v9, v10.

aswaterman, 2020-08-14:
“For any vrgather instruction, the destination vector register group cannot
overlap with the source vector register groups” — so in case 2, vs1 can’t
be v11, either.

Rolfy360, 2020-08-14:
I think I should post this issue to spike, because current spike validates the relations among rd, rs2, rs1 using 

if (__builtin_expect(!(insn.rd () != insn.rs2 () && insn.rd () != insn.rs1 ()), 0))
  throw trap_illegal_instruction (0);

rd and rs1 should use is_overlapped(.) to check the overlap condition.

Rolfy360, 2020-08-14:
post this issue to spike repository.

closed 2020-08-14 by Rolfy360