riscvarchive/riscv-v-spec #562: rsqrt example code looks incorrect

JamesKenneyImperas, 2020-08-27:
I think that function `rsqrt` in `recip.cc` might be wrong. For a negative infinity argument, it returns a _zero of the same sign_ but I think this should be a _QNaN result setting invalid operation_. Can you please confirm?

JamesKenneyImperas, 2020-08-27:
ignore that - my mistake

JamesKenneyImperas, 2020-08-27:
Sorry for thrashing, but I _do_ in fact think it is incorrect at line 121, so reopening this issue.

aswaterman, 2020-08-27:
I thought I fixed this weeks ago, but maybe I forgot to push it.  Sanity-check my fix, please.

JamesKenneyImperas, 2020-08-27:
I think that looks right. I fixed it in my version by changing line 121 only to:

`    if (!sign && exp == ((1UL<<E)-1) && sig == 0) {
`

but your change looks equivalent.

Thanks,

James.

aswaterman, 2020-08-27:
Ah yes, that would've worked, too.  I already did it the other way, so... :)

closed 2020-08-27 by aswaterman