riscvarchive/riscv-v-spec #567: Allow different src/dst LMUL for vrgather?

knightsifive, 2020-09-07:
An increasingly common use-case for `vrgather.vi` is in applications like normalizing a vector, where we perform a reduction into element 0 of a (_single_) vector register and then "splat" its result onto another vector register _group_. The obvious alternative, using two `vmv` instructions, loses decoupling between scalar and vector pipelines. Other memory-system-based alternatives have their own obvious disadvantages.

My reading of the spec is that it intends for a `vrgather`'s source and destination vector register groups to be the same size (i.e., same LMUL). As such, this effectively constrains which register is used for the reduction result. For example, if we want to splat onto an LMUL=8 group, we must ensure that the reduction result is in `v0`, `v8`, `v16`, or `v24`, perhaps incurring an extra `vmv`, increasing register pressure and complicating other scheduling.

This isn't a bug in the spec, but it is an inconvenience. 

To keep my criticism constructive, please let me suggest a mitigation, allowing a different effective LMUL for `vrgather`'s source operand. To follow the register grouping rules, this EMUL could be derived as min(LMUL, 2^ctz(`vs1`)), e.g., if LMUL = 4 and `vs1` = `v18` then EMUL = 2. I am happy to submit a PR if there is interest, but wanted to sound out the idea here first.

aswaterman, 2020-09-08:
My initial take is that the cost of an extra `vmv` relative to the cost of a reduction+splat is sufficiently low that this is more of an annoyance than a significant performance artifact.  It does complicate scheduling, but the worst case is that it exposes the vmv->splat latency, which is typically quite small relative to the occupancy of the reduction and large-LMUL splat.  In this case, there's no extra register pressure, since the splat can use the same source and destination.

Perhaps a better solution is to use multiple splats: e.g., if the reduction result went into v4 but you want to do an LMUL=8 splat, you can do two LMUL=4 splats instead (might need to use the form of vsetvli that sets vl=VLMAX).  This has the downside of increased code size vs. your proposed solution, but otherwise it should perform the same: the total occupancy is the same, and there is no additional scheduling constraint.  (Actually, there is more scheduling flexibility.)

knightsifive, 2020-09-08:
> In this case, there's no extra register pressure, since the splat can use the same source and destination.

> For any vrgather instruction, the destination vector register group cannot overlap with the source vector register groups, otherwise an illegal instruction exception is raised.

What am I misunderstanding?

aswaterman, 2020-09-08:
> What am I misunderstanding?

Nothing, I was mistaken.  But I don't think this has any impact on the alternative (multiple-splat) scheme.

knightsifive, 2020-09-08:
I was initially put off by the excessive splatter but I think I can get used to it.

closed 2020-09-08 by knightsifive