riscvarchive/riscv-v-spec #568: Is there a way to disable the vector unit?

brucehoult, 2020-09-09:
Early versions of the draft spec specified a way for a program to indicate that it wasn't using the vector unit any more but current drafts don't.

This seems like a useful thing for several reasons:

1) not having to save and restore the vector registers on task switch for a program that has used the vecvtor unit but is not currently using it.

2) making it easier to share one vector unit between several CPU cores. I seem to recall there was a presentation at the RISC-V Global Summit by someone who has done this, though I can't recall who.

I can't find any commentary or git issue explaining why vdisable was removed.

The history I can find is below.

The first commit of files in this repo included in v-spec.adoc the line:

`A pseudo-instruction vdisable is mapped to vconfig with all immediate bits 0 (vregmax=0, vemaxw=0).`

Then in "c88970a organizational changes":

`A pseudo-instruction `vdisable` is mapped to `vconfig` with all immediate bits `1` (`vregmax=0`, `vl=0`).`

Then in "ee694d9 Describe vconfig encoding scheme":

`A pseudo-instruction `vdisable` is mapped to `vconfig` with immediate bits `0xff`, resulting in `vregmax = 0` and `vl = 0`.`

Later in "4db6709 some cleanup" this was changed to:

`A pseudo-instruction `vdisable` is mapped to `vconfig` with immediate bits `0x7f`, resulting in `vregmax = 0` and `vl = 0`.`

Then in "24d41e2 Overhaul vregcfg/vfastcfg scheme":

`A pseudo-instruction `vdisable` is mapped to `vconfig` with immediate bits `0x01`, resulting in `vl = 0`.`

The last commit with this text was "2b15889 Saving changes before replacing 0.5-draft" and there is no mention in "55fb32a Replaced spec with new draft." from Nov28 2018 or anything more recent.

aswaterman, 2020-09-09:
For determinism, vdisable needs to clear the architectural state.  Since we got rid of mandatory tail-zeroing, implementations generally won't have a mechanism to quickly clear the VRF.  So, to avoid a bunch of vdisable-specific datapath and complexity, most implementations would probably choose to implement vdisable by actually clearing the state elements that comprise the VRF, which comes an energy cost, and for some implementations a time cost (especially if the vector unit is reused soon after a vdisable).

The upshot is we can't reasonably recommend that compilers automatically insert a vdisable in the epilogue of a function that uses the vector unit: over time, the overhead could easily exceed the context-switch savings.  vdisable would thus probably be relegated to situational invocation by intrinsic, calling its utility into question.

The proposed system-call ABI provides an out: whenever a context switch occurs as the result of a blocking system call, the OS can choose to reset VS to Initial.  After such an event, preemptive context switches will also avoid the overhead of saving the state, until the next time the vector unit is used.  This strategy covers the two important bases of vector-heavy processes (which wouldn't benefit from vdisable anyway) and syscall-heavy processes (which can freely use vector implementations of C string functions without fear of blowing up the context switch time).  (SVE employs a similar strategy.)

brucehoult, 2020-09-09:
So if you use the vector unit the full state will be saved on a context switch, but lazily reloaded?

aswaterman, 2020-09-09:
These optimizations are related but separately motivated.  The lazy context-switch code is informed by sstatus.VS.  The optimization I described judiciously clears sstatus.VS, so that there's no need to save (or later restore) the state.

brucehoult, 2020-09-10:
Ahh. I guess that's reasonable. Any function call is entitled to nuke all of the vector registers, so a system call certainly is.

However this doesn't address the case of embedded systems. Ok, I guess they can just explicitly manipulate mstatus if they want to.

kasanovic, 2021-03-19:
Considered closed.

closed 2021-03-19 by kasanovic