riscvarchive/riscv-v-spec #571: Behavior of vfredmax.vs/vfredmin.vs/vfwredosum.vs when none vector elements are active?

zhongqibiao, 2020-09-23:
In section 15.3.1. Vector Ordered Single-Width Floating-Point Sum Reduction, there is a note says "If no elements are active, no additions are performed, so the scalar in vs1[0] is simply copied to the destination register, without canonicalizing NaN values and without setting any exception flags. This behavior preserves the handling of NaNs, exceptions, and rounding when autovectorizing a scalar summation loop. "
So what about the vfredmax.vs/vfredmin.vs/vfwredosum.vs instructions? Do they have the same behavior as vfredosum.vs when no elements are active?

aswaterman, 2020-09-23:
The earlier sentence "Inactive elements from the source vector register group are excluded from the reduction, but the scalar operand is always included regardless of the mask values" answers this question implicitly: the instruction performs a reduction on a sequence consisting only of vs1[0], i.e., the result is vs1[0].

zhongqibiao, 2020-09-24:
@aswaterman I think the sentence you mentioned does not explain clearly what we should do when execute a float reduction instruction without any active element and vs1[0] is NaN. Both vfredosum.vs and vfredsum.vs sections have specific description about that corner case. The vfredosum.vs will copy vs1[0] to vd[0] directly without  setting any exception flag, but the behavior of vfredosum.vs is implementation-defined. So in my view, different float reduction instructions may have different behavior when no element is active and vs1[0] is NaN. As there is not any description in the vfredmax.vs/vfredmin.vs/vfwredosum.vs/vfwredsum.vs sections, I can't say the spec is very clear.
I think the reasonable rule is that vfredmax.vs/vfredmin.vs/vfwredosum.vs instructions behave same as vfredosum.vs and the vfwredsum.vs instruction behaves same as vfredsum.vs when no element is active and vs1[0] is NaN, right?

aswaterman, 2020-09-24:
I do think it follows from the sentence I quoted; the redosum instructions are special cases that override the default behavior.  But I agree it wouldn't hurt to add a NOTE.

commit 29aa87c2 (2020-09-24): Clarify NaN handling for fully inactive vfredmin/max/vfwredosum

Resolves #571

No need for normative text; these NOTEs follow from the intro to the
reductions section.
closed 2020-09-24 by aswaterman