riscvarchive/riscv-v-spec #575: Are the reserved bits of vtype WARL?

respasa, 2020-09-29:
In this file: 

https://github.com/riscv/riscv-v-spec/blob/master/vtype-format.adoc 

bits XLEN-2:8 are described as "Reserved (write 0)".
 
Should this be interpreted as:
a) WARL (where legal would be zeros)?
b) Trap if attempting to write a non-zero value?
c) Other?

Whatever the answer, can we clarify in the doc?

thanks

roger.

kasanovic, 2020-10-23:
While allowed to trap on reserved values, we don't want to mandate data-dependent traps.
Implementation can set vill if not a supported configuration.

vsetvli instructions have a 11b value written to vtype (sign bit of 12b imm field selects immediate/register forms of vsetvl) .
For future extension, we could view vtype as an 11b CSR, with additional CSRs added for more bits later.

riking, 2020-11-15:
Note that as you cannot directly write to vtype, a WARL designation would be moot.

kasanovic, 2021-06-07:
Changed to remove "write 0" in figure.  The current text has correct spec that unknown vtype setting will clear low order bits in vtype.

commit 59235fb9 (2021-06-07): Removed the "write 0" request for vtype.  Can only be written via  vsetvl and that definition gives correct behavior.
Closes #575.
closed 2021-06-07 by kasanovic