riscvarchive/riscv-v-spec #578: provide a note that emphasizes that implementations are free to trap on vsetvl{i}

David-Horner, 2020-10-02:
I'm OK with vill set if vtail is unsupported. However, I believe we should provide a note that emphasizes that implementations are free to trap on vsetvl{i}

    on any specific combination of vtype values
    to trap conditionally (typically with a CSR configuration setting)
    recommend to not trap on implementation supported vtype settings
    and highlight the benefit of immediate trapping (either explicitly itemizing and explaining or referencing another document). [ some of those being: definitive identification of instruction, allowing precise determination of arguments, which in turn allows setting an equivalent vtype (in a typical case substitute tailun for tailag)]

see:
https://github.com/riscv/riscv-v-spec/issues/367#issuecomment-582715647
this was initially raised in the context of the agnostic tail debate

kasanovic, 2020-10-20:
In general, implementations are free to trap on any instruction or any value in general.  Adding a special note here would cause readers to question if trapping is allowed for every other instruction where a note is not present.

David-Horner, 2020-10-20:
@kasanovic  Adding a special note here would cause readers to question if trapping is allowed for every other instruction where a note is not present.

I suspect most readers will not be so confused. It is hardware engineers and EE devlopers that care about such things.

By not adding the note many more will be unaware of the consequences of not trapping.
What does a hardware engineer care about EE and user code challenges?
It is the classic rhetorical question.

kasanovic, 2020-10-20:
OK - we can add a note that if writes of illegal vtype values are not trapped, alternative vector designs can not be emulated as the use of vill is too late to take action.

I will say that the community of hardware engineers are very much the ones that worry when things are not consistent between different instruction descriptions.  I think the important difference here is the defined setting of illegal bit versus undefined reserved behavior on an illegal value.

commit 8e3d3374 (2020-10-26): Add note that implementations may trap when illegal values written to vtype.
Closes #578.
closed 2020-10-26 by kasanovic