riscvarchive/riscv-v-spec #584: `npm build` fails due to lack of pushd

timsifive, 2020-10-09:
When I try to build the spec on Ubuntu 20, I get:
```
tnewsome@compy-linux:~/SiFive/riscv-v-spec$ npm run build

> riscv-v-spec@0.1.0 build /home/tnewsome/SiFive/riscv-v-spec
> mkdir -p public && pushd public && ../node_modules/.bin/datasheet -i ../v-spec.adoc -o riscv-v-spec.pdf && popd

sh: 1: pushd: not found
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! riscv-v-spec@0.1.0 build: `mkdir -p public && pushd public && ../node_modules/.bin/datasheet -i ../v-spec.adoc -o riscv-v-spec.pdf && popd`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the riscv-v-spec@0.1.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/tnewsome/.npm/_logs/2020-10-09T16_21_20_651Z-debug.log
```

[2020-10-09T16_21_20_651Z-debug.log](https://github.com/riscv/riscv-v-spec/files/5356237/2020-10-09T16_21_20_651Z-debug.log)

Presumably this is because my /bin/sh doesn't have pushd, and on some systems pushd does work because /bin/sh is really /bin/bash.

drom, 2020-10-09:
do you run `bash`?

David-Horner, 2020-10-09:
I had the same problem this morning.

> When I try to build the spec on Ubuntu 20, I get:
>
> tnewsome@compy-linux:~/SiFive/riscv-v-spec$ npm run build
>
>
> sh: 1: pushd: not found
> npm ERR! code ELIFECYCLE
> npm ERR! syscall spawn
> npm ERR! file [...]
I copied the string out of the json and ran in the public directory to make
it work.

drom, 2020-10-09:
@timsifive @David-Horner could you try #585 ?

timsifive, 2020-10-09:
> do you run bash?

Yes:
```
tnewsome@compy-linux:~$ echo $SHELL
/bin/bash
```

PR #585 (2020-10-09): fixes #584 ; `npm build` fails due to lack of pushd
fixes #584
commit a545a670 (2020-10-09): fixes #584 ; `npm build` fails due to lack of pushd (#585)
closed 2020-10-09 by aswaterman