riscvarchive/riscv-v-spec #591: Whole-register operations with non-zero vstart

JamesKenneyImperas, 2020-10-26:
I can see the sections for whole-register operations are changing. I'm not sure implications when vstart>0 are correctly reflected in the current descriptions. Can you please clarify? For example section 7.9 says:

> The load instructions have an EEW encoded in the mew and width fields following the pattern of regular unit-stride loads. Because in-register byte layouts are identical to in-memory byte layouts, these instructions all operate the same as moving vectors of bytes with EEW=8, regardless of actual EEW encoding. Pseudo-instructions are provide for whole register load instructions that correspond to EEW=8.

If vstart=0, I can see that the load/store would apply to the whole register, and therefore the effect of the instruction is independent of EEW. But what happens if vstart>0? At what position in the register does the load/store start? Later, there is a statement:

> The instructions operate with an evl=nf*VLEN/EEW, regardless of current settings in vtype and vl. The usual property that no elements are written if vstart ≥ vl does not apply to these instructions. Instead, no elements are written if vstart ≥ evl.

If the EEW in the instruction is just a hint, the EEW here is presumably 8; is that right?

Apologies if this isn't clear; I think what I'm trying to say is can the behavior of these instructions when vstart>0 be more clearly defined?

Thanks.

aswaterman, 2020-10-26:
I think vstart is supposed to be in units of elements (i.e. EEW)--consistent with the fact that implementations are allowed to require EEW alignment.  So EEW isn't merely a hint.

JamesKenneyImperas, 2020-10-27:
Assuming the encoded EEW is not just a hint, I think the words here definitely need some further rework.

One possible clarification could be to describe the register state after the point when an instruction such as `vl1re64.v v3, (a0)` traps part way through because of a page fault, making explicit:
1. What element size was used for each load (the encoded `EEW` or `SEW_MIN`);
2. What value is in `vstart` if the fault occurs after (say) 12 bytes have been loaded, if alignment to `EEW` isn't implemented.

kasanovic, 2021-06-07:
I think this was clarified by https://github.com/riscv/riscv-v-spec/commit/c841730fe683243531a0d5ffa087b584683d6e57

closed 2021-06-07 by kasanovic

David-Horner, 2021-06-07:
I don't see c841730 addressing this:

> What value is in vstart if the fault occurs after (say) 12 bytes have been loaded, if alignment to EEW isn't implemented.

Nor does it appear to be answered throughout the rest of the whole register section.

In general it would be decent to allow the originator opportunity to respond before the issue is closed, even to close it themselves.

In this specific case, I believe the issue is not fully addressed, and so it should not have been closed.

JamesKenneyImperas, 2021-06-07:
Hello David,

I was also unclear about what was intended here, and had an exchange with Andrew Waterman in the comments section of the original pull request he made - did any of that get copied to you? I don't seem to be able to find the thread now that the pull request has been approved for some reason (I'm not an expert GitHub user, so I don't know why).

My summary would be that originally these whole-register instructions were defined to operate with an EEW of 8, but over time this has changed so that the encoded EEW is respected (not just a hint) and specifies the element size for transfer, just like normal load/store instructions. Unfortunately this did not happen as a single atomic change, so the specification has for a while has been unclear - some text has implied that the encoded EEW is used, and some that it is not and that an EEW of 8 is used. Given this evolution, I can't see how alignment to EEW can _ever_ be unsupported now, unless it is also allowed for normal loads and stores, but perhaps I'm still not understanding the intent fully.

I looked at the vector working group minutes and issues, but couldn't find any reference to this apparent change in semantics. However, I'm only a distant observer of the specification, so this probably doesn't mean much.

James.

kasanovic, 2021-06-07:
The change to respecting EEW, not only treating as hint, was discussed at length in several TG meetings and should be visible in minutes and/or TG google doc notes.  Minutes of 2020/11/06 captures the exception-handling behavior discussion and this basically moved treatment of encoded EEW from hint to architectural.  The rest of the behavior follows from that transition.

The new text does not call out any difference from unit-stride loads and stores which also have explicit EEW, as vstart behavior is same as for those.  The difference due to statically encoded evl is described.

If base address is misaligned, then first element is misaligned and it follows from other parts of spec that vstart must be zero on exception.

This is a longer and separate discussion, but issue open/close feature on GitHub is used as part of maintainers' workflow not only to record objections/concerns.   The issue content (including the thread on the pull request), are still visible after being closed (click on "closed" word in issue or pull request listing or follow link in the emails generated by GitHub), and as seen in action here, the originator of an issue that was closed by another is free to comment and ask for issue to be reopened.   Waiting for issue originator to agree to close issue (or conversely allowing them to close an issue before it was dealt with), are both problematic for workflow.

David-Horner, 2021-06-07:
The workflow deficiency of "closed" is also apparent. Closed encompasses 
more than is intended, even by your reckoning.

The deficiency is substantially with Github. It does not provide a 
robust work flow infrastructure.

 From your example, you cannot [easily and readily] stop someone who 
opened the issue from closing it, even though that messes up your 
workflow management.

...

As you say, it is a longer and and separate discussion that I hope is 
addressed in more than this issue item.

True. But I am sure you understand that the usability and utility of 
Github features is not my expressed concern.

obviously I am aware. But again I expect you are aware of the messaging 
and psychological impediment presented by premature closing of issues.

This depends upon a level of trust that may not have been earned, and 
for some has been lost already by the premature closing.

Similarly, capitalize first letters of sentences is problematic for some 
typists.

But in deference to others, for clarity and convention, people make 
efforts/allowances to make the whole better.