riscvarchive/riscv-v-spec #593: Load masks

daweili1226, 2020-11-03:
Hi all,

If I store vector masks as bits in memory, for example,
VL = 8
mask = 10010010
The mask is a byte long,  then how do I load it as a mask?

aswaterman, 2020-11-03:
Regular vector instructions targeting the mask register, e.g.,

```
li t0, 1
vsetvli x0, t1, e8,m1,ta,ma
vle8.v v0, (addr)```

daweili1226, 2020-11-03:
Hi Andrew, 

If I understand correctly, the above code works if masks are aligned as bytes in memory, but to further save memory, I would like to store masks as bits, which means for vl = 16, I can use 2 bytes to encode the mask for an entire vector register, then can I still efficiently load the mask?

knightsifive, 2020-11-03:
I'm not clear on your question. The `vl == 16` generalization of @aswaterman's code is the following:
```
li t0, 2
vsetvli x0, t1, e8,m1,ta,ma
vle8.v v0, (addr)
```
`addr` need not be 2-byte aligned (i.e., even).

However, you could also implement this code as follows:
```
li t0, 1
vsetvli x0, t1, e16,m1,ta,ma
vle16.v v0, (addr)
```
In this case, portable software must have `addr` be 2-byte aligned.

This is analogous to the scalar memory operations, e.g., `lb` and `lh`. See: https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#9-vector-memory-alignment-constraints

daweili1226, 2020-11-04:
Now I understand the concept,
```
li t0, 2
vsetvli x0, t0, e8,m1,ta,ma
vle8.v v0, (addr)
```
By this, I will have 16-bit mask located in v0, then I will need an additional vsetvli instruction to fetch my original settings back, right?
Do we have a plan to add load mask instructions from either memory or GPRs?

kito-cheng, 2020-11-04:
> Do we have a plan to add load mask instructions from either memory or GPRs?

I think load mask from GPR by `vmv.v.x` if mask is less than XLEN bits, and combine with `vslide1up` to move any bits from GPRs, SLEN is gone.

knightsifive, 2020-11-04:
Agree with @kito-cheng.

Also, I don't think there is a plan to add "load mask" instructions, since it can be simulated so easily. And, in my experience on SiFive's implementations, `vsetvli` is very cheap: I've wasted much time optimizing away `vsetvli`s, seeing no discernible improvements.

It's likely I don't completely understand your application, or your concern, so please feel free to provide a longer code example.

daweili1226, 2020-11-06:
Just seeking the most efficient way to load masks because this operation is inside my hottest code pattern.

closed 2021-01-29 by kasanovic