riscvarchive/riscv-v-spec #594: Could "vsetvl" skip the vstart reseting?

JerryShih, 2020-11-04:
In https://github.com/riscv/riscv-v-spec/issues/127, it says that "vsetvl" will also reset the vstart to 0.

If I try to call the vleXXX with the non-zero vstart value, I should put the vstart setting code after the "vsetvl" instruction.

e.g.
```
vsetvli XXX
csrw vstart, t0
vleXXX
```

The following code is not work:
```
csrw vstart, t0
vsetvli XXX // This vsetvli will reset the vstart to 0. The csrw before this line is not work as expected.
vleXXX
```

aswaterman, 2020-11-04:
Right- you should place the csrw vstart after the vsetvl. I don’t see any
reason to redefine these instructions.

JerryShih, 2020-11-05:
There are some limitations for current rvv intrinsic implementation. The vleXXX intrinsic will always comes with a preposing vsetvl instruction. Then, the vstart will be reset to 0.

The spec says that changing vstart manually is not a good practice. And some platforms might not support the non-zero vstart value. So, close this issue here.

closed 2020-11-05 by JerryShih

aswaterman, 2020-11-06:
Right, userspace code should generally not be adjusting vstart.