riscvarchive/riscv-v-spec #595: Strided stores with rs1=x0

kasanovic, 2020-11-09:
In our earlier TG discussion in 9/18 meeting, we were in favor of
allowing vector strided load instructions with rs1=x0 to perform fewer
memory accesses than the number of active elements.  This allows
higher-performing splats of a scalar memory value into a vector.

In writing this up, I inadvertently made this true for stores too.
But on review, I can't see a reason to not also allow strided stores
(which are now unordered), to also perform fewer memory operations (in
effect, picking a random active element to write back).  The behavior
is indistinguishable from a possible legal execution of prior scheme,
and has potential niche use of storing element value to memory when it
is known all elements have same value.

https://github.com/riscv/riscv-v-spec/commit/398d453e3592efbac77cc8f6658009759901185a#diff-ea57dd7a8daf0aa62f553688c1970c8e6608945d25597f8661c5ea6670fb509c

I suppose we could also reserve the encoding with strided stores of
rs1=x0, but this would add some asymmetry.  Software could then get a
similar effect by settng vl=1 before the store.

knightsifive, 2020-11-09:
I'm fine with this behavior, understanding the typos as fixed in https://github.com/riscv/riscv-v-spec/commit/6ae609416e0900bf96fb9daa6c83d701ac833a9a, although I admit I don't yet have a real-world example of this niche use case.

knightsifive, 2020-11-09:
As @aswaterman and I just discussed, there are many numerical algorithms where we, say, perform a dot product (reduction) and then normalize a vector by the dot product as well as (b) write the dot product into an output matrix. Moving the dot product to the scalar register file loses decoupling, making it possibly worthwhile to splat and use `.vv` arithmetic. To then write the dot product to memory we'd need a `vsetvli` (AVL = 1), and perhaps a second one to "undo" the first. This proposed instruction would avoid that overhead.

From a software portability perspective, I think I'd tend to write the code with the extra `vsetvli`(s) since I'd expect that overhead to be much less than that of an unoptimized `vsse`.

kasanovic, 2020-11-10:
Yes - for performance portability,  cannot rely on hardware optimizing away all the stores when rs2=x0, so explicit vl=1 is correct approach for portable optimized software to take.   I imagine nearly all implementations will ignore rs2 zeroness and just perform strided stores an element at a time under assumption software will explicitly ask for a single store when that is desired behavior.  I believe this new definition does not actually change the set of allowable behaviors of strided stores from previous one,  just makes it symmetric with strided load definitions.

kasanovic, 2021-06-07:
There does not seem to be any reason to restrict allowed store behaviors here, so closing the issue without changing spec.

closed 2021-06-07 by kasanovic