riscvarchive/riscv-v-spec #597: Use lg2(2*SEW) shift amount for vssrl and vssra

HanKuanChen, 2020-11-13:
For `vsll`, `vsrl` and `vsra`, the output is 0 when shift amount equals to SEW. Use lg2(SEW) to determine shift amount is enough.
However, because fixed-point has rounding mode, the output **may not be 0** for `vssrl` and `vssra`.
For example, the output is 1 when `vxrm` is 0 with a data 128 and a shift amount `8`. It would be 0 if `vxrm` is 2.
To do a fixed-point shift (and shift amount may equal to SEW), the code would be
```
# v1 is data
# v2 is shift
# a0 is vl

# the following code is for vxrm 0
# use other algorithm if vxrm is not 0
vsetvli x0, a0, e32, m1
vmv.v.i v3, 1
vadd.vi v4, v2, -1
vsll.vv v4, v3, v4
vand.vv v4, v1, v4
vmseq.vi v0, v4, 0	# set if rounding is not required
vmerge.vvm v4, v3, 0, v0
li a1, 32
vmsne.vx v0, v2, a1
vssra.vv v4, v1, v2, v0.t

# v4 is result
```
to handle this problem. But the problem can be solved easily if lg2(2*SEW) is used to determine shift amount.

knightsifive, 2020-11-17:
I think I disagree with the hypothesis, 

> For vsll, vsrl and vsra, the output is 0 when shift amount equals to SEW.

The single-width shifts say that

> The low lg2(SEW) bits of the vector or scalar shift amount value are used

In other words, you can have a shift of 0, 1, ..., SEW-1, but not equal to SEW.

This matches the base integer ISAs, which only allow shifts of 0, 1, ..., XLEN-1, i.e., the range of a lg(XLEN)-bit unsigned immediate. Like many other RISC-V idiosyncrasies, this appears to have its roots in the C language:

> The behavior is undefined if `rhs` is negative or is greater or equal the number of bits in the promoted `lhs`. 

https://en.cppreference.com/w/c/language/operator_arithmetic#Shift_operators

HanKuanChen, 2020-11-17:
> In other words, you can have a shift of 0, 1, ..., SEW-1, but not equal to SEW.

Yes. The v-spec does not support a SEW shift. But if users want to simulate the behavior when shift amount may equal to SEW, they can do by themselves easily
```
# v1 is data
# v2 is shift
# a0 is vl

# for vsll
vsetvli x0, a0, e32, m1
vsll.vv v1, v1, v2
vmseq.vi v0, v2, 8
vmerge.vxm v3, v1, x0, v0

# for vsrl
vsetvli x0, a0, e32, m1
vsrl.vv v1, v1, v2
vmseq.vi v0, v2, 8
vmerge.vxm v3, v1, x0, v0

# for vsra
vsetvli x0, a0, e32, m1
vmseq.vi v0, v2, 8
vsra.vv v1, v1, 1, v0.t
vsra.vv v3, v1, v2

# v3 is result
```

But for `vssrl` and `vssra` is hard, **because it has rounding modes**. Users need to implement 3 additional algorithms (when `vxrm` is 2, it behaves like `vsrl` or `vsra`) for `vssrl` and `vssra` respectively (total 6 algorithms). Each of the algorithms is a little complicate.

> Like many other RISC-V idiosyncrasies, this appears to have its roots in the C language:

But C language does not support fixed-point operations.

aswaterman, 2020-11-17:
It's possible to do this without different algorithms per rounding mode.  For shamt >= SEW, you can first perform a right-shift by floor(shamt/2) using the ROD rounding mode, then perform a right-shift by ceil(shamt/2) using the originally requested rounding mode.

(I think this algorithm only works for shamt >= 4, so it would suffice to only engage this algorithm when shamt >= SEW, and use the single instruction when shamt < SEW.)

aswaterman, 2020-11-17:
Also, this complexity is only necessary for SEW=ELEN; for SEW<ELEN, it's faster to widen then use vnsrl/vnsra.  I imagine the SEW<64 case is much more common in practice.