riscvarchive/riscv-v-spec #598: provide a cononical instruction for vector instructions synch points - for imprecise trap management

David-Horner, 2020-11-13:
The spec allows imprecise trap behaviour.
Effective management of imprecise traps is greatly assisted by providing synch points.
Consider debug which is simplified considerably by checking the vector state when there are no vector instructions in-flight, that is partially completed [even if the re-execute would suffice].
A synch point provision will be provided for this purpose by any manufacturer that provides imprecise traps. However, to provide a canonical approach by a specific instruction will allow portability to complying implementations.
In #364 I proposed only consecutive [in the instruction stream] vector instructions could overlap processing. Thus any non-vector instruction [which would not include vsetvl and its varients] would be a synch instruction.
However, future designs could be much more aggressive and still be manageable.
Providing this canonical instruction would by be honoured by all imprecise trap approaches.

I don't have a specific proposal, but I am inclined towards selecting one of the fence variants. 
What of 
VFENCE.V same as FENCE.I except bits 12-14 are 010 instead of 001.

I'm inclined to green field within either FENCE or FENCE.I as I see no need for multiple variants of VFENCE.V [and, obviously, no need for all variants for both FENCE and FENCE.I]. But perhaps I am being short sighted.

kasanovic, 2020-12-18:
Providing a trap fence should probably be more global than just vector instructions, though it might have variants that allowed selection of a subset of traps.

kasanovic, 2021-06-07:
Labeling as post v1.0, though is really a broader issue than vector extension.