riscvarchive/riscv-v-spec #604: Problems in using "Vector Integer Arithmetic Instructions" to calculate fixed-point calculations

lzwButton, 2020-11-30:
In calculation, I have to see fixed-point as integer to calculate because the lack of fixed-point instructions. For most situation they are the same but they have different behavior facing overflow.

For integer it only saves the lower bits and fixed-point has saturation. 

Therefore, in some fixed-point functions I have to extend the elements to deal with overflow. Which I could use LMUL=8 to read in one loop, now I can only use LMUL=4 to have protect bits and I also have to clip it down.

My question is that is it possible to make integer instructions have saturation feature. Like there is a bit to control whether it has saturation feature or not. This looks like more flexible.

By the way, do you have some other ways to deal with the overflow in fixed-point function other than extend elements to set protect bits?

knightsifive, 2020-11-30:
I'm confused by this question. Currently, the V-extension provides a very small set of fixed-point instructions:
https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#vector-fixed-point-arithmetic-instructions
Some of these perform saturation.

Can you please provide more details on the algorithm that you are trying to implement that is not well addressed by this set of functions?

lzwButton, 2020-12-01:
I think this situation mostly happened in "redsum" together with "mul". Even though I used widened instruction, there is still a chance of overflow. And because it is a integer instruction, I can not deal with it when overflow happened. So I use "wredsum" together with "wmul" and clip the result which required extra resources and more cycles. 

I wonder if there will be a fixed-point version of "redsum".

knightsifive, 2020-12-01:
Are you trying to implement a gemmlop-style quad-widening GEMM? If so, I'd use quad-widening FMAs 
https://github.com/riscv/riscv-v-spec/blob/master/quad-mul.adoc
and then clip the output matrix register tile before writing it back to memory.

Unfortunately, the quad-widening FMAs won't be part of the 1.0 release.

Personally I'm in favor of removing *all* the fixed-point instructions from 1.0, but that's another topic..

lzwButton, 2020-12-01:
Perhaps that is one way to optimize it. Still, it requires clip.

I thought vector is born to battle with SIMD. Ideally, it do behave better than SIMD. But most SIMD are used in DSP or NN which calculate with fixed-point inputs. Without fixed-point instructions perhaps vector may not be better than SIMD. 

Actually, I am agree to you about removing all the fixed-point instructions from 1.0. To be honest, it is really makes me confuse sometime to choose whether the integer instructions or fixed-point instructions. But anyway, I think the integer instructions should have a saturation version.

ZPedro, 2020-12-01:
It is my experience that instructions that saturate rather than wrap, while punctually useful, are insufficient in the general case because they don't compose as well. For instance, with vsadd.vv you can perform perform (a + b) with saturation only at the last step, however you cannot do even just (a + b + c) with saturation only at the last step, because intermediate saturations may change the final result, even after that result is subject to saturation. It can be proven that intermediates would require more than n bits in order to keep enough information so the final result is correct, even after saturation.

The same goes for instance for averaging add, which works for 2 inputs (and could be tolerated for specific situations such as 4 inputs if you're willing to tolerate the loss of accuracy from the intermediate roundings), but completely breaks down for something as simple as averaging 3 inputs with equal weight.

As a result, in the general case there is no substitute to performing your fixed-points computations with an element width wider than your outputs, and clip the computation outcome to your output width, with the clipping instruction performing most if not all the saturation in the process.

lzwButton, 2020-12-02:
I understand what you said. What I thought is that every addition is saturated. In this case, even overflow happens in a+b, it still can keep most information. Because the overflow will happen anyway, why bother to use protection bits. The saved bits can extend the elements that calculated one loop. 

Perhaps I got it wrong, I think "redsum" with saturation means every addition is saturated, rather than the last step. Just tell me if I got it wrong. Maybe it can not designed like that.

In conclusion, if every addition are saturated, there is no need to have protection bits. If the overflow happened, that means these data has no use. So simply set it to max value is enough, I think. If I need the overflowed data, I would not clip it down and directly write it back.

ZPedro, 2020-12-02:
I admit I have less experience with redsum and other horizontal operations, but aren't they typically performed once at the end of processing, and thus out of the hot path? And so any additional complexity at that point in your code would be amortized anyway?

lzwButton, 2020-12-14:
After learned for a long time, I finally got it clearly. What I need is simply a dot instruction. My purpose is to write a loop to fulfill mult accumulate which returns a scalar. This issue is raised because I think it can be optimized with saturation.  Actually, saturation may not be that necessarily for me, If I can use dot instructions.

closed 2020-12-14 by lzwButton