riscvarchive/riscv-v-spec #610: Will there be dot instructions

lzwButton, 2020-12-14:
To fulfill mult-accumulate and return a scalar, I tried mac combined with redsum. Actually, if there are dot instructions, all these extra cycles and codes will never be used. 

Then I noticed that dot instructions is clearly announced that it would not be in v1.0. It is a pretty useful instruction in matrix calculation.

So, my question is that will there still be dot instructions in future. If so, will there be a date, like in v2.0 or else. Will it be added in extension.

aswaterman, 2020-12-15:
Yes, the proposal is that dot instructions will be part of the Zvediv extension, finishing defining which is future work.

lzwButton, 2020-12-15:
Thank you for your reply. According to your reply, I may say that it would not be sure how dot instruction actually is. Like will it have vwdot and will there be saturation.

closed 2020-12-15 by lzwButton

aswaterman, 2020-12-15:
The details are TBD, but I would expect we'd follow the existing spec: vdot[u] will wrap, whereas vsdot[u], if provided, will saturate.

lzwButton, 2020-12-15:
Got it! Personally, it would be great to have vsdot and it can be easily used in fixed-point application.

knightsifive, 2020-12-15:
More generally, we don't offer a (single-width or widening) saturating FMA and reduction. These are all useful in "quantized" neural network algorithms.

fanghuaqi, 2020-12-15:
> 
> 
> More generally, we don't offer a (single-width or widening) saturating FMA and reduction. These are all useful in "quantized" neural network algorithms.

Hi @knightsifive , I don't quite understand what do you mean by not offer a  (single-width or widening) saturating FMA and reduction. Could you help to make it clear. 

From the conversation, if I want to use `vdot` or `vsdot`, could be ready in v-spec 1.0, even in `Zvediv` extension is ok for us, it will make neural network algorithms easy to be optimized.

knightsifive, 2020-12-15:
The current draft (confusingly called v1.0) does _not_ include the following:

1. saturating integer FMA
2. widening saturating integer FMA
3. saturating integer reduction
4. widening saturating integer reduction
5. integer dot product
6. widening integer dot product
7. saturating integer dot product
8. widening saturating integer dot product

... among many other instructions you might want for quantized NNs.

The Zvediv draft, which has no clear schedule/timeline --- it it certainly won't be part of v1.0 --- includes (5) and (6), but none of the others.

There are a lot of omissions in the fixed-point support. I think there is a good argument to have a separate AI/NN (sub-) extension, rather than trying to refocus V-extension at this specific application domain. However, this would probably have to start with a proposal at RISC-V Int'l.

fanghuaqi, 2020-12-15:
Hi @knightsifive , thank you for your clarification, it make me clear now. 

I am looking forward to the AI/NN (sub-) extension or Zvediv extension, it will of course speed up NN application, currently v-spec 1.0, we can't get a lot speed up in such application case.