riscvarchive/riscv-v-spec #611: [Vector-Float] NX/NV Bit Setting on VFCVT

CohenSteed, 2020-12-14:
TL; DR -
**Does VFCVT xu.f.v set NV, NX, or both for negative float values?**

I couldn't find a discussion target for Floating Point questions. This is tied to Vector implementations of Chapter 14 (Vector Float operations).
Conversion from a float (any size) to an UNSIGNED integer (any size) must adhere to the table presented in the relevant floating-point specification, or, by undefined extension, whatever interpretation of 16-bit floating point from IEEE is used. (V, Chapter 14).
The F/D specifications give a table of what to do with values outside of min/max range. This includes float values outside of an integer range. Negative float values are inherently outside of unsigned integer range, and the table reflects this.

However, the specification intentionally and directly states the assertion of the INVALID (NV) flag in these cases, in addition to how to handle the value itself. Note that setting the INVALID (NV) flag is a departure from IEEE-754-2008, the foundation for the floating point operations and definition.

According to IEEE-754-2008, for an operation where the real value cannot be precisely represented, the INEXACT (NX) flag is set. This is easy to spot on every system, ever: x86, ARM, even SPIKE itself behaves this way. The INVALID (NV) flag is not set in cases of conversion from negative float value to unsigned integer. However, the RISC-V F/D spec specifically requires it.

At the same time, no mention of the INEXACT (NX) is used, in this context, in the RISC-V specification.

Should the result flag, for conversion of a negative float to an unsigned integer, be:
- NX
- NV
- NV and NX
?

CohenSteed, 2020-12-14:
If this question is better answered in another forum, please direct me to that location.

knightsifive, 2020-12-14:
My understanding is that RISC-V floating-point is fully IEEE-754 compliant.

> According to IEEE-754-2008, for an operation where the real value cannot be precisely represented, the INEXACT (NX) flag is set. This is easy to spot on every system, ever: x86, ARM, even SPIKE [...]

My reading of IEEE 754-2008 is that we are supposed to set NV for most float-to-unsigned conversions of negative inputs:

> For operations producing no result in floating-point format, the operations that signal the invalid operation exception are [...]     conversion of a floating-point number to an integer format, when the source is NaN, infinity, or a value that would convert to an integer outside the range of the result format under the applicable rounding attribute

The only exception I can see is negative floats with magnitude sufficiently small to get rounded to zero.

Regardless, NX should be set too, as you explained.

CohenSteed, 2020-12-14:
Thanks for the response.
I guess I am seeing ambiguity in how this is written. Is compliant meaning RISC-V mimics the entire asserted state - in other words, all flag states are asserted or cleared explicitly - or only that RISC-V asserts the same flags, but also possibly more?

If it's the latter - that RISC-V sets more flags than IEEE-754 specifies - then the tables in the F/D spec make sense. If it's the former, where RISC-V precisely adheres to the flag state per IEEE-754, then the tables in the F/D spec don't seem to make sense to me. Perhaps I'm confused on how the sources read?

knightsifive, 2020-12-14:
My understanding is that RISC-V only asserts the flags than IEEE 754 specifies. Could you please clarify which "tables in the F/D spec" are suggesting otherwise?

Are you perhaps looking at the tables in Vol. 1 Ch. 14 (Memory Consistency)? If so, please keep in mind those tables do not imply that *all* the listed flags are necessarily asserted simultaneously.

CohenSteed, 2020-12-14:
Absolutely - clarification is in order.

RISC-V, Vol. 1, Ch. 11, Sect. 7
> If the rounded result is not representable in the destination format, it is clipped to the nearest value and the invalid flag is set. Table 11.4 gives the range of valid inputs for FCVT.int.S and the behavior for invalid inputs.

Given that "out-of-range negative input" is explicitly a row in Table 11.4, and the FCVT.WU.S and FCVT.LU.S columns are for unsigned integers, the notion of any negative floating point value (it's out of range, as a negative) will return 0 and the INVALID flag is required to be set.

Assert: This is a clear indication that any negative float will convert to unsigned integer value of 0, with the NV flag set.
(Minor aside - does this include negative zero? This is also unclear, since negative zero is not considered a negative value for comparisons, but is considered less-than positive zero for comparisons)

IEEE-754 (2008), 5.8 "Details of conversions from floating-point to integer formats" (emphasis mine)
> When a NaN or infinite operand cannot be represented in the destination format and this cannot otherwise be indicated, the invalid operation exception shall be signaled. When a numeric operand would convert to an integer outside the range of the destination format, the invalid operation exception shall be signaled **if this situation cannot otherwise be indicated.**
> When the value of the conversion operation’s result differs from its operand value, yet is representable in the destination format, some conversion operations are specified below to signal the inexact exception and others to not signal the inexact exception.
...
IEEE-754 (2008), 7.2 "Invalid operation"
> For operations producing no result in floating-point format, the operations that signal the invalid operation exception are:
i) conversion of a floating-point number to an integer format, when the source is NaN, infinity, or a value that would convert to an integer outside the range of the result format under the applicable rounding attribute

Assert: This lines up cleanly with the RISC-V specification, and INVALID should be signaled in these cases according to both specifications.

What about INEXACT?
> Unless stated otherwise, if the rounded result of an operation is inexact—that is, it differs from what would have been computed were both exponent range and precision unbounded—then the inexact exception shall be signaled. The rounded or overflowed result shall be delivered to the destination.

Assert: I clearly have more homework to do. These look like they mimic each other appropriately, and specific cases in the specification should be established. It is much more likely that this is correct as-is, and I need to better understand specific cases.

Thanks again. I'll leave this open for final comments.

knightsifive, 2020-12-14:
Not sure if we are in (dis)agreement about one minor point, so please let me reiterate:

Section 11.7 says:
> If the rounded result is not representable in the destination format, it is clipped to the nearest value and the invalid flag is set.  Table 11.4 gives the range of valid inputs for FCVT.int.S and the behavior for invalid inputs.

Then, the first row of Table 11.4 clarifies that, for conversions-to-unsigned, the

> Minimum valid input (after rounding)

is "0".

With this in mind, my reading is that the subsequent "out of range negative input" row corresponds to "negative finite floating point values that, when rounded (a la _rm_) to an integer, remain negative."

I guess our disagreement (if any) concerns the fact that when a negative floating point number is rounded to "zero", it will actually result in negative zero. My interpretation is that negative zero is a nonnegative integer, so is thus not out of range. However, I'm not exactly sure how to back this up using the (RISC-V or IEEE 754) specs.

CohenSteed, 2020-12-14:
The minor point plays out, using a SoftFloat test library, for what that's worth (and it should be worth a lot). Tiny negative floating numbers convert to integer value 0, with NX set.
This makes sense.

There are other differences I am investigating on a chart of "interesting integer values" that scale from Canonical NaN to Positive Infinity, Bigger-than-maxpos-Int, and so on. Once my results are compiled, I'll share the table here.

I feel like I opened a can of worms, and the current outcome I expect is that it might be possible to come to a bad conclusion on the RISC-V implementation of F/D IEEE-754 with the way the chapter is written. I'm not saying the intent is wrong, but that the wording can lead to an awkward result.

Given how well-defined and published the F/D portions of the spec are, it might not be so easy to rewrite those sections. Thoughts?

aswaterman, 2020-12-14:
@knightsifive it's a bit of a weak argument, but you can rationalize the fact that negative zero should round to integer 0 with no exceptions on the basis that they're supposed to compare as equal.

knightsifive, 2020-12-14:
If you spot unclear or ambiguous wording in the spec, please PR your suggestions to `riscv-isa-manual`.

CohenSteed, 2020-12-15:
Looking at the source documents, the discrepancy is cleared up. I'm going to close the issue.
As for negative zero, I agree with @aswaterman : if an item resolves to negative zero (based on basic signage), it is further converted to positive zero in all but a narrowly defined set of cases where reporting negative zero reports information that would otherwise be lost. These cases are enumerated in IEEE-754 (2008). Conversion to Integer is not one of those cases, so the answer for unsigned integer result is clear:
- round a negative number to the appropriate integer (rounding-mode).
- if the value is negative zero, adjust to positive zero, set NX, and report.
- if the value is negative non-zero, report value per RISC-V F/D spec, set NV, report.

closed 2020-12-15 by CohenSteed