riscvarchive/riscv-v-spec #615: Minimum value of VLEN when XLEN=32

MarkHillHuawei, 2020-12-17:
I've been reviewing the rationale for making the minimum value of VLEN 128 bits as discussed in "Mandate VLEN>=128 for application processor profile. Issue #533"

I think it is entirely reasonable for RV64 machines to set the minimum as twice the width of a GPR. However, in the highly PPA sensitive embedded space, enforcing an RV32 machine to implement a minimum register file size of  32*128-bits risks making the solution uncompetitive with respect to other commercial offerings which, for example, are already sharing the register file with the FPU.  We could in theory switch to the packed extension at the low end but from an ecosystem perspective it is much more attractive to use a single SIMD architecture and perform the PPA trade-offs in the microarchitecture.

Would it be possible to the set the minimum value to 2*XLEN?

David-Horner, 2020-12-17:
The short answer is yes. For embeded or any particular profile minimum VLEN
can be 2 * XLEN.

What is apparently not yet clear is that the default [working] profile is
LINUX based applications. The min VLEN in particular is to allow code
portability in that ecosystem.

Specifications for other domains/profiles/platforms can vary. Embeded, in
particular, is less portability constrained/sensative so the minimum there
can certainly be different. Different tradeoffs.

brucehoult, 2021-04-11:
I note that ARM MVE (aka Helium) specifies 8 vector registers of 128 bits each.

This is the same configuration and total size as if RVV on 32 bit embedded allows VLEN = XLEN and software uses LMUL=4.

I've looked at #533 and I think the example given there for requiring VLEN to be greater than XLEN doesn't make sense (and I've commented on that there).

kasanovic, 2021-06-07:
Hopefully clarified with https://github.com/riscv/riscv-v-spec/commit/808a6f83b72d92757ef4c93fcdf076ed99bbecae

closed 2021-06-07 by kasanovic