riscvarchive/riscv-v-spec #619: Clarify rules on vector/scalar when XLEN<SEW

frasercrmck, 2020-12-21:
I'd like to request that the behaviour of vector/scalar operations when XLEN<SEW is clarified a little. Namely, it's not immediately obvious whether the register and immediate behaviours are always in agreement:

> For integer operations, the scalar can be a 5-bit immediate encoded
> in the `rs1` field.  The value is sign-extended to SEW bits, unless
> otherwise specified.  . For integer operations, the scalar can be
> taken from the scalar `x` register specified by `rs1`.  [...] 
If XLEN<SEW, the value from the `x` register is sign-extended to SEW bits.
 
> NOTE: Some instructions _zero_-extend the 5-bit immediate, and denote this
> by naming the immediate `uimm` in the assembly syntax.

In the cases that the immediate value is zero-extended, is the corresponding register value zero-extended or does it continue to be sign-extended?

David-Horner, 2020-12-22:
Whereas unsigned might be much more valuable for short immediate values, the same is not necessarily true for register based variable values.

In my opinion there does not need to be a one-to-one correlation for instructions of zero-extended immediate and unsigned register values.

knightsifive, 2020-12-22:
As it says here, it is "sign-extended unless specified otherwise". A few months ago @aswaterman and I went through the spec to confirm that all "otherwise" cases were well defined. (You can search the spec for "uimm", I see just 21 instances in the current version.)

aswaterman, 2020-12-22:
I think a literal reading of the current spec is unambiguous. Maybe some of the specified behavior is surprising, leading to some doubts about the intentions.

frasercrmck, 2020-12-23:
> I think a literal reading of the current spec is unambiguous. Maybe some of the specified behavior is surprising, leading to some doubts about the intentions.

Thanks for your responses. Yeah, I think you've hit the crux of the matter. I agree that it is unambiguous when read literally and so that's how we did it in LLVM. I'm not here to argue there *should* be a one-to-one correlation, just for clarification. I raise the issue because in my experience there is sometimes a disconnect between how a specification is written literally and the actual intention behind it. If you can stop people second-guessing what they're reading then that's all the better. But I'm also happy for the issue to be dropped at this point.

kasanovic, 2021-06-07:
Closing issue.

closed 2021-06-07 by kasanovic