riscvarchive/riscv-v-spec #622: About vl when AVL < (2 * VLMAX)

luyahan, 2021-01-04:
```
1.vl = AVL if AVL ≤ VLMAX
2.ceil(AVL / 2) ≤ vl ≤ VLMAX if AVL < (2 * VLMAX)
3.vl = VLMAX if AVL ≥ (2 * VLMAX)
```
But i don't understand what value be set to vl when VLMAX < AVL < 2*VLMAX.

I note:
```
For example, this permits an implementation to set vl = ceil(AVL / 2) for VLMAX < AVL < 2*VLMAX
```

Should 6.2 be fixed ?
```
1.vl = AVL if AVL ≤ VLMAX
2.vl = ceil(AVL / 2)  if VLMAX < AVL < 2*VLMAX
3.vl = VLMAX if AVL ≥ (2 * VLMAX)
```

nick-knight, 2021-01-04:
In this case, the value of VL is implementation defined. The intent is that an implementation can choose VL < VLMAX to balance the work across the last two iterations of a strip-mined loop. (Maybe somebody else can explain why that's beneficial..) However, I expect most implementations will pick VL = VLMAX.

knightsifive, 2021-01-04:
(Reposting from correct account.) In this case, the value of VL is implementation defined. The intent is that an implementation can choose VL < VLMAX to balance the work across the last two iterations of a strip-mined loop. (Maybe somebody else can explain why that's beneficial..) However, I expect most implementations will pick VL = VLMAX.

luyahan, 2021-01-05:
> (Reposting from correct account.) In this case, the value of VL is implementation defined. The intent is that an implementation can choose VL < VLMAX to balance the work across the last two [...]

Thx!

closed 2021-01-05 by luyahan