riscvarchive/riscv-v-spec #623: IEEE754 compliance: round/ceil

jan-wassenberg, 2021-01-11:
Quoting from IEEE Std 754-2008 5.3.1.0:

> Implementations shall provide the following homogeneous general-computational operations for all
supported arithmetic formats; these operations shall not propagate non-canonical results. Their destination
format is indicated as sourceFormat:
```
sourceFormat roundToIntegralTiesToEven(source)
sourceFormat roundToIntegralTiesToAway(source)
sourceFormat roundToIntegralTowardZero(source)
sourceFormat roundToIntegralTowardPositive(source)
sourceFormat roundToIntegralTowardNegative(source)
```
These round/floor etc functions are [widely used](https://github.com/search?l=C&q=_mm256_round_ps&type=Code). I didn't see an issue discussing this yet. Should these functions be added? As things stand, we'd have to convert to integer and back.

kasanovic, 2021-01-30:
The sequence developed in TG meeting was:
```
    vfcvt.rtz.x.f.v v8, v4     # Round to integer of same width (could use frm)
    vfcvt.f.x.v v8, v8         # Round back to float of same width
    vfabs.f.f.v v12, v4        # Get magnitude of original value.
    vmfgt.vf v0, v4, f0        # Is large integer already? f0 has threshold.
    vmerge.vvm v4, v8, v4, v0  # Leave alone if already an integer
```
But not clear that this is correct wrt IEEE spec.  Group was to discuss relative importance of this operation.

aswaterman, 2021-01-30:
The three corner cases this doesn't handle are preservation of -0.0 inputs, propagation of NaN inputs, and avoiding signaling the invalid exception on qNaN inputs.  I believe the following code sequence is correct and really not all that much worse.  (It's inspired by the equivalent RV scalar sequence in glibc: https://github.com/bminor/glibc/blob/master/sysdeps/riscv/rvf/s_rintf.c)

    # mag = fabsf(x);
    # res = x * 1.0f;  // canonicalize NaNs; no effect on others
    # if (!isnanf(x))
    #   if (mag < float(1L << 24))
    #     res = copysignf(float(int(x)), x);
    # return res;

    vfabs.v     v8, v4             # Compute magnitude
    vmfeq.vv    v0, v4, v4         # Mask active for non-NaN inputs
    vfmul.vf    v12, v4, f1        # Canonicalize possible NaN inputs
    vmflt.vf    v0, v8, f2, v0.t   # Mask active below threshold
    vfcvt.x.f.v v8, v4, v0.t       # Round to integer of same width
    vfcvt.f.x.v v8, v8, v0.t       # Round back to float
    vfsgnj.vv   v12, v8, v4, v0.t  # Handle -0.0 case

jan-wassenberg, 2021-02-03:
Hi @aswaterman , thanks for pointing out the corner cases. I agree fixing -0 is affordable and we get NaN for free from the vfmerge, right?

This is now implemented in [Highway](https://github.com/google/highway/blob/master/hwy/ops/rvv-inl.h#L1523). I see our CI/CD is spamming this issue with comments, sorry about that :)

It seems to me we'd be fine with emulating like this instead of a native instruction, does anyone else have any thoughts on that?

aswaterman, 2021-02-03:
@jan-wassenberg There's no merge, just masking.  But specifically, NaNs are handled as follows: 1) the `vfmul` canonicalizes NaN inputs; 2) the `vmfeq` clears the mask for NaN inputs (and raises invalid on signaling NaNs); 3) the `vfcvt.f.x.v` does not overwrite the canonicalized NaN outputs because they're masked off.

I agree that emulating this way is suitable, unless a compelling performance case for adding new instructions is made sometime in the future (not for 1.0 :))

jan-wassenberg, 2021-02-04:
@aswaterman right, thanks for clarifying. In the Highway implementation, we're skipping the vfmul and just letting an `abs(in) < MaxMantissa` comparison return false for NaN.

Come to think of it, I'm moving up the CopySign so we don't return -NaN.

I think we can close this unless anyone has any other comments.

kasanovic, 2021-06-07:
Closing issue.

closed 2021-06-07 by kasanovic

aswaterman, 2022-02-11:
Right, the code we provided correctly implements `rintf`, not `nearbyintf`, though the latter can be emulated by saving and restoring NX.  As usual, we'd need a quantitative justification from realistic workloads to add new instructions, which hasn't been presented.