riscvarchive/riscv-v-spec #626: vsetvl[i] is available or not when enable only zvamo or zvlsseg

kito-cheng, 2021-01-22:
IIUC, we can only enable zvamo or zvlsseg in theory, but it seems like need some fundamental operation to work like `vsetvl[i]` instruction and whole register move/load/store for spilling and register move.

So how do you think about to made `vsetvl[i]` instruction and whole register move/load/store are available if any zv* extension is enabled?

aswaterman, 2021-01-22:
Zvamo or Zvlsseg is totally useless without vsetvl[i], so the dependence makes sense.

But, I don't think allowing these extensions by themselves makes sense, either.  Shouldn't we define a Zvbase (which includes vsetvl[i] and other basic instructions), and say that Zvamo etc. depend on Zvbase?

knightsifive, 2021-01-22:
I agree. The spec occasionally mentions a "base" V-extension: it would be nice to finally define what this means.  For example,

> 7.8. Vector Load/Store Segment Instructions
> This instruction subset is given the ISA string name Zvlsseg.
> Note | This set of instructions is included in the base "V" extension used for the Unix profile.

:thinking:

kito-cheng, 2021-01-22:
> I agree. The spec occasionally mentions a "base" V-extension: it would be nice to finally define what this means. For example,
>> Note | This set of instructions is included in the base "V" extension used for the Unix profile.

@knightsifive yeah, that's also kind of vague to me, especially the term of Unix profile` :P

kasanovic, 2021-06-07:
Removing AMO extension and integrating vlsseg into bases has made this point mostly moot for now https://github.com/riscv/riscv-v-spec/commit/808a6f83b72d92757ef4c93fcdf076ed99bbecae https://github.com/riscv/riscv-v-spec/commit/a8098bd1ac8cab5cd8dbd9056e33ae1a4dc17d33

Later, for readability and because there different vector bases, you would normally list the vector base V or Zve32x and also list Zvamo or other vector extensions.

closed 2021-06-07 by kasanovic