riscvarchive/riscv-v-spec #630: vsetvl instruction decode clarification

JamesKenneyImperas, 2021-02-03:
The decode pattern given for the `vsetvl` instruction has the value `0x1000` specified for the penultimate field. Can you explain what this means? I was expecting all zeroes here, which is what it was in the "old" instruction format description.

(I haven't looked at this since everything was converted to the new format a few months ago.)

```
{reg: [
  {bits: 7,  name: 0x57, attr: 'vsetvl'},
  {bits: 5,  name: 'rd', type: 4},
  {bits: 3,  name: 7},
  {bits: 5,  name: 'rs1', type: 4},
  {bits: 5,  name: 'rs2', type: 4},
  {bits: 6,  name: 0x1000},
  {bits: 1,  name: 1},
]}
```

aswaterman, 2021-02-03:
I have no idea what it means, but it renders as all zeros:

[image]

JamesKenneyImperas, 2021-02-04:
A stupid question: how do I see the rendered result? It would by nice if the links from the specification could show this instead of the raw form.

aswaterman, 2021-02-04:
Take a look at the README.  It requires some setup.

JamesKenneyImperas, 2021-02-04:
ok, thanks.

closed 2021-02-04 by JamesKenneyImperas

David-Horner, 2021-02-04:
Even though it can be generated locally, having a 'properly' rendered pdf
on the site for major updates is as helpful in riscv-v-spec as in
riscv-isa-manual.

My experience has been that I got different renderings. Perhaps due to
slightly different versions of software, perhaps configuration defaults.
But an official  reference would avoid spurious reports of errors AND
provide an authoratative document.

JamesKenneyImperas, 2021-02-04:
Yes David, I completely agree