riscvarchive/riscv-v-spec #635: vsetivli constant vl interpretation

JamesKenneyImperas, 2021-02-04:
The new `vsetivli` instruction encodes the required `vl` as a constant in the range 0...31. I'm wondering if the instruction could be made more useful if the constant field was interpreted differently. For example, I can see no use for `vl` of 0, but can imagine that `vl` of 32 would often be desired.

Perhaps the `uimm` field could be reinterpreted to provide the 32 most useful values, based on some analysis of existing software.

JamesKenneyImperas, 2021-02-04:
I see from reading the meeting minutes that this was discussed there. I'll assume this has been fully considered and close this issue.

closed 2021-02-04 by JamesKenneyImperas

jnk0le, 2021-06-27:
There seems to be no concrete conclusion in the minutes.

>There was discussion on whether uimm=00000 should represent 32 or be
reserved.  32 is more useful, but adds a little complexity to
hardware.