riscvarchive/riscv-v-spec #640: `vrgatherei16` can't be freely used in case of unconstrained VLMAX

jnk0le, 2021-02-06:
continuation on #520, similarly applies also to `vid` instruction.

Compilers and programmers can't simply assume that exceeding 2^16 VLMAX is not gonna happen, otherwise software will break on future processors. (and definitely will in configurable emulators)
It means that produced code must always assume worst case, being 2^(XLEN-1), which requires extra boilerplate code to provide artificial constrain on AVL, serving separate codepaths or literally processing 8bit data at XLEN SEW.

Possible solutions are:

a) constrain worst case vlmax to 2^16. RVV++ can't lift this restriction, otherwise legacy software will break.
b) provide `vrgatherei32` and constrain worst case vlmax to 2^32 (31?)
c) provide vrgatherei{XLEN} (or 64?, RV128 will half the available lmul) so at least data can be processed at native size (probably not better from boilerplate)

Even if constraint will be left to a profiles, (b) and (c) options require extra instructions

jnk0le, 2021-02-16:
There is one more possible approach:

d) VLEN/VLMAX is not restricted "by spec/profile", but the 11th bit (`zimm[10]`) in `vtypei`/`vtype` will reduce allocatable `vl` by `vsetvli`/`vsetvl` to <= 2^16 for 16bit SEW at `lmul`==8 (2^17 for 8bit SEW at `lmul`==8). `vsetivli` has limited range by 5bit `uimm` so it needs only to "default" to one of the options. That "default" could be `1` bit that is placed in same position as `zimm[10]` of `vsetvli`, so the decoders can handle both (11 and 10 bit) `vtypei` constants with same circuit.

Of course this wastes a whole bit in precious `vtypei` space and is also gonna raise issues with access to "inaccesible" upper part by slides etc.

kasanovic, 2021-02-22:
Group agreed that profile or platform would be responsible for setting the VLMAX limit, with current guidance being that  profiles should bound at 64K or below.

kasanovic, 2021-03-12:
A realization is that this VLMAX limit has to be part of the ISA spec, not only a  profile-imposed limit.   It is not possible to increase the limit in a backwards-compatible way (e.g., to allow libraries compiled under v1.0 model to be used in a >v1.0 machine with longer vector registers), without adding some new forms of instruction to set vl allowing larger VLMAX.

So the base vector extension will impose this limit.  The form of the limit is most easily expressed as a limit on VLEN, rather than VLMAX.  Hence, VLEN <= 65,536 is part of spec now.

jnk0le, 2021-03-12:
> VLEN <= 65,536 is part of spec now.

I think that can be extended to 131072
- SEW==8 doesn't care about `VLMAX`
- SEW==16 gets exactly 65,536 `VLMAX`

kasanovic, 2021-03-12:
@jnk0le  I don't understand this comment. 

Why does SEW=8 not care about VLMAX?

If VLEN is 128Kb, LMUL=8 and SEW=8, vsetvl can set vl to 128K.  Vector instructions are not the only place where the index size constraint would be applied.  E.g., scalar code might be doing the indexing and want to know 16b is large enough.

jnk0le, 2021-03-16:
When needed to be idexed by 16bit SEW the lmul has to be reduced and thus VLMAX.

>Vector instructions are not the only place where the index size constraint would be applied. E.g., scalar code might be doing the indexing and want to know 16b is large enough.

Is this even representable in C, that the code is doing stripmining but 17bit vlmax is a problem?

The only affected instance I found is the slide amount in slide instructions. vmv and vfmv acces only element 0.
I'm not sure that can affect even dedicated code, considering the fact that scalar registers are at least 32bit wide.

kasanovic, 2021-03-19:
Code might store the VLMAX vector to memory and then index using scalar addressing.

commit 2054e4a5 (2021-06-08): Added upper limit of 64Kib on VLEN.
Closes #640.
closed 2021-06-08 by kasanovic