riscvarchive/riscv-v-spec #641: is it possible to add outer product calc as an arithmetic instruction

julian-xinchi, 2021-02-07:
https://en.wikipedia.org/wiki/Outer_product

David-Horner, 2021-02-08:
Highly unlikely for v1.0.

The original vision for RISCV vector did propose matrix multiplication, including outer product.
However, the Vector TG decided that vision was too visionary.
The result is a comprehensive but limited set of augmented 32bit instructions.
Direct matrix operations will be a later release, planned to use 64bit instruction encoding.

V1.0 does provide features that support efficient matrix multiplication operation code sequences.
Notably these include:
- stride load/store to efficiently load the columns of arrays.
- shift and move under mask to duplicate the row data by the number of columns.
--  whole register move [copy] between physical vectors registers for use when LMUL>1
- For full matrix [not just outer product] there are [3 argument] fused multiply add variants.

julian-xinchi, 2021-02-08:
> Highly unlikely for v1.0.
>
> The original vision for RISCV vector did propose matrix multiplication, including outer product.
> However, the Vector TG decided that vision was too visionary.
> The result [...]

Sweet! Thanks for your reply. Maybe i more expect the version after v1.0 :-)