riscvarchive/riscv-v-spec #644: Need clarification regarding vl preserving vsetvl{i}

g-jose, 2021-02-18:
In section 6.1 the vector spec states the following regarding vsetvl{i} instructions when rd=x0, rs1=x0:

> Use of the instruction with a new SEW/LMUL ratio that would result in a change of VLMAX is reserved. Implementations may
set vill in this case.

Does this mean that implementations:
(i) can choose to either trap or set vill on such a usage of vsetvl{i}, or
(ii) should trap on such an instruction but optionally set vill bit, or
(iii) can choose to neither trap nor set vill bit but modify vl silently?

kasanovic, 2021-02-19:
(i) (with assumption most implementations would prefer to just set vill)

not (ii), reserved  does not imply "must trap"

(iii) would currently be a non-conforming extension, but in future a new standard extension might have this behavior.

daweili1226, 2021-03-03:
In section 12.6 the vector spec states bit shift instructions without mentioning the operations and the role of the operands, people therefore have to "guess" that vs2 might be the source and "vs1, rs1, imm" are shift amounts.  Taking bit shift instructions as an example, I would suggest we add missing descriptions.

kasanovic, 2021-05-13:
@daweili1226 The shift instruction description was updated in commit https://github.com/riscv/riscv-v-spec/commit/942320c54d8c3808702cd4f2a89655082163d968

In future, please create a separate issue for each separate topic.

closed 2021-06-09 by kasanovic