riscvarchive/riscv-v-spec #648: vmsbf.m, vmsif.m and vmsof.m register overlap constraints

JamesKenneyImperas, 2021-03-03:
On May 14th 2020, overlap constraints for `vmsbf.m`, `vmsif.m` and `vmsof.m` instructions were tightened to disallow destination overlap with either source or mask; previously, there was no such constraint on any of these instructions. It is usually (fairly) obvious to me what hazard the overlap constraints are there to prevent, but in this case it isn't.

I thought it was worth checking that this change was required and isn't in fact introducing an unnecessary restriction. This was during the major mask register format restructuring, so there was a lot of upheaval.

Thanks.

aswaterman, 2021-03-03:
I'm not the author, but here's my rationalization for the design:

If we had been OK with these instructions being uninterruptible, we could've permitted overlap.

AFAICS, restarting these instructions in the middle is possible but adds extra complexity: you need to scan the destination register to see if you already wrote a 1.

To avoid the more complex restart mechanism but still allow interrupts, we need to restart from the beginning, implying that we can't overwrite the destination.

JamesKenneyImperas, 2021-03-03:
Yes, I think that explains it: I had overlooked the interaction with interrupts. Perhaps the restriction for these three instructions could be updated to say:

The destination register cannot overlap the source register and, if masked, cannot overlap the mask register ('v0'). _This is required for correct restart behavior after an interrupt_.

Thanks.

closed 2021-03-03 by JamesKenneyImperas