riscvarchive/riscv-v-spec #649: Whether the implementation description of wfredsum is a requirement or suggestion?

zhongqibiao, 2021-03-07:
In section 15.3.2, it says "The implementation can ...", the word "can" makes me feel it's a suggestion. But the discussion in issue #386 makes me feel the description in spec is a requirement.

So I want to make sure whether the description of wfredsum is a requirement that every implementation must follow? Is it acceptable that the reduction tree of a implementation calculate more than two operands each node, or always use RTZ instead of the dynamic rounding mode? 

I think the order and implementation difference of floating point sum reduction instruction will affect whether it can get the exact result and what result value it will get when raising the inexact flag. So maybe the basic requirement of the wfredsum instruciont is it should produce exact reduction sum result as using infinite exponent range and precision, otherwise the inexact flag must be raised. 

I am also interested in when and how the vfredsum instruction will be used in software program. In my opinion, this instruction is not friendly to portable code. Any information about how to use it will help understand the requirements of the instruction. Thanks.

aswaterman, 2021-03-08:
The use of "can" is unfortunate here; it is meant to be a requirement.  I will fix the text.

The intent is _not_ to mandate an exact result, but merely to admit that as one possibility.  It is too expensive to mandate that hardware.

Yes, the instruction is inherently problematic for portability and should only be used when different summation results on different implementations are acceptable.

closed 2021-03-08 by aswaterman

jan-wassenberg, 2021-03-08:
@aswaterman I observe that vfredmin of all-NaN is FLT_MAX instead of NaN. Is that desirable? If so, it doesn't seem to be permitted by the current spec?

aswaterman, 2021-03-08:
I don't think that's right.  vfredmin of {NaN, NaN} is NaN.  vfredmin of {x, NaN, NaN} is x.  This is consistent with a scalar loop using the fmin instruction.

Unlike vfredsum, vfredmin doesn't give the implementation any wiggle room, because it's cheap to implement in a manner that's consistent with the scalar loop.

knightsifive, 2021-03-08:
Additionally, unlike `fadd`, `fmin` is associative (as is IEEE `minNum`, as of IEEE 754-2019 [[Chen 2017](https://grouper.ieee.org/groups/msc/ANSI_IEEE-Std-754-2019/background/minNum_maxNum_Removal_Demotion_v3.pdf)]). So it's unclear what wiggle room an implementation could be reasonably allowed.

jan-wassenberg, 2021-03-09:
@aswaterman @knightsifive 
Thanks for looking into this. I agree that the IEEE 754-2018 behavior is desirable, perhaps we're looking at a qemu bug - if so, where is that best reported?

The following code works on ARMv8 (vminnm), x86 (minps) but not riscv64-unknown-elf-g++ (SiFive GCC 10.1.0-2020.08.2) 10.1.0 / qemu-riscv64 version 5.1.0 (v5.0.0-2894-g73537b968d):
```
template <typename TF>
bool IsNaN(TF f) {
  MakeUnsigned<TF> bits;
  memcpy(&bits, &f, sizeof(TF));
  bits += bits;
  bits >>= 1;  // clear sign bit
  // NaN if all exponent bits are set and the mantissa is not zero.
  return bits > ExponentMask<decltype(bits)>();
}

template <class D, class V>
void AssertNaN(const D d, const V v, const char* file, int line) {
  using T = TFromD<D>;
  const T lane = GetLane(v);
  if (!IsNaN(lane)) {
    const std::string type_name = TypeName(T(), Lanes(d));
    MakeUnsigned<T> bits;
    memcpy(&bits, &lane, sizeof(T));
    // RVV lacks PRIu64, so use size_t; double will be truncated on 32-bit.
    Abort(file, line, "Expected %s NaN, got %E (%zu)", type_name.c_str(), lane,
          size_t(bits));
  }
}

struct TestFullNaN {
  template <class T, class D>
  void operator()(T /*unused*/, D d) {
    const auto v1 = Set(d, T(Unpredictable1()));
    const auto nan = IfThenElse(Eq(v1, Set(d, T(1))), NaN(d), v1);

// Reduction (pending clarification on RVV)
#if HWY_TARGET != HWY_RVV
    HWY_ASSERT_NAN(d, MinOfLanes(nan));  // = vfredmin
#endif
  }
};
```

aswaterman, 2021-03-09:
Yeah, if a min-reduction of all NaNs is returning Inf, it's an implementation bug.  These days, it's actually best to file RISC-V QEMU bugs in the official manner (https://www.qemu.org/contribute/report-a-bug/).

jan-wassenberg, 2021-03-09:
Got it, thanks, will do.

pliaotw, 2021-03-11:
@chihminchao @frankchang0125 FYI

jan-wassenberg, 2021-03-11:
Thank you @pliaotw , I understand this has already been addressed.

commit 5538929e (2021-03-08): s/can/must/

Resolves #649