riscvarchive/riscv-v-spec #650: Revisit agnosticism

knightsifive, 2021-03-07:
From the [definition of agnosticism](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#333-vector-tail-agnostic-and-vector-mask-agnostic-vta-and-vma):

> The value of all 1s instead of all 0s was chosen for the overwrite value to discourage software developers from depending on the value written. 

I still don't feel that this is a sound technical justification.

There are applications in linear algebra computations for tail-zeroing computations, like zero-padding matrices when repacking into a block layout. Programmers who work in these domains are typically tuning for particular microarchitectures, and portability isn't a real concern to them. Such programmers could take advantage of microarchitectures that implement `ta` with zeroing, rather than the far less useful current definition ("perhaps 1, perhaps undisturbed").

I propose relaxing the current definition to allow writing zero.

David-Horner, 2021-03-07:
> From the definition of agnosticism
> <https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#333-vector-tail-agnostic-and-vector-mask-agnostic-vta-and-vma>:
>
>     The value of all 1s instead of [...]
I was keen on a more poisonous pill. So I agree about lack of sound 
technical justification, but for the exact opposite cause.
> There are applications in linear algebra computations for tail-zeroing
> computations, like zero-padding matrices when repacking into a block
> layout. Programmers who work in these domains are [...]
There is no reason that a "non-conforming" implementation cannot zero-pad.

Nor that a conforming implementation cannot have a non-conforming mode 
where by zeros are padded.

Nothing prohibits implementors from "extensions".

> I propose relaxing the current definition to allow writing zero.
I can see a future extension that provides zero-fill, but it is less 
than compelling to provide it now when the actual value is unknown.

We have all agreed that agnostic can provide performance improvement for 
envisioned implementations.

There are less tangible benefits that agnostic encourages, such as 
portability.

In this case we addressed the elephant in the room, and even though it 
is a compromise,

it is compatible with a future "agnostic definition" that is less 
constrained.

In other words more relaxed.

knightsifive, 2021-03-07:
Hi David, 

Thanks for your response. I think I agree with most of your comments. But I still don't see a convincing technical argument for allowing tail-asserting (or "-uniting"?) --- while precluding tail-zeroing -- as a (conformant) implementation of tail agnosticism.

I don't even fully buy the subjective, psychological-conditioning argument given in the spec: what's to keep a programmer from now expecting tail-asserting behavior? Plus there's already many ways of writing nonportable code.

From a software perspective, saying "agnostic" means "I don't care what certain elements hold after the instruction", not, "I don't care, provided no zeros overwrite ones". 

From a hardware perspective, 
> each destination element can be either left undisturbed or overwritten with 1s, in any combination, and the pattern of undisturbed or overwritten with 1s is not required to be deterministic when the instruction is executed with the same inputs.

So "determinism" (for verification, e.g.) cannot be a valid argument. And I don't see this constraint simplifying implementations either.

David-Horner, 2021-03-08:
>
> Hi David,
>
> Thanks for your response. I think I agree with most of your comments. 
> But I still don't see a convincing technical argument for allowing 
> tail-asserting (or "-uniting"?) --- while precluding tail-zeroing -- 
> as a (conformant) implementation of tail agnosticism.
I agree that there is no technical argument for the current 
specification. I think agnostic should mean exactly that, any value 
could be presented.

However, we do want restrictions on what can be returned.

In particular we do not want to allow bleed through from other 
processes. We want to ensure side channel attacks are not possible.

So we already have a constraint. It is a small step from that to 
specific restrictions that are known to be safe.

Zero would be safe, as would 1s, or tail elements from destination 
register, or from either source register or the results of the operation 
on the operands.

> I don't even fully buy the subjective, psychological-conditioning 
> argument given in the spec: what's to keep a programmer from now 
> expecting tail-asserting behavior?
I am concerned about agnostic being misunderstood and programmers assume 
their machines implementation is the same as all others.
>
> Plus there's already many ways of writing nonportable code.
Only too true. I think it is worthwhile us making efforts to help avoid 
this problem.

If we can eliminate or drastically reduce the problem, then that is a 
great accomplishment.

> From a software perspective, saying "agnostic" means "I don't care 
> what certain elements hold after the instruction",
true
>
> not, "I don't care, provided no zeros overwrite ones".
But agnostic can also include that version.
> From a hardware perspective,
>
>     each destination element can be either left undisturbed or
>     overwritten with 1s, in any combination, and the pattern of
>     undisturbed or overwritten with 1s is [...]
Verification does not need this degree of "determinism".

The tail will be deterministically compatible with the spec, even when 
the bit sequence generated will not be identical to the previous execution.

>  And I don't see this constraint simplifying implementations either.
It isn't simplifying per se. In some cases it may make implementations 
more complex.

However, there is a class of designs that benefit from not having to 
copy over the tail unchanged elements.

aswaterman, 2021-03-08:
The premise in the original post won't always hold up.  Part of the reason for allowing nondeterminism even for the same inputs to allow threads to migrate between cores that implement this property differently.  So, determining that you're currently running on a core that zeros the tail isn't a guarantee that you can rely on tail zeroing for the duration of the process' execution.

David-Horner, 2021-03-08:
>
> The premise in the original post won't always hold up.
original post:

>> Programmers who work in these domains are ... tuning for particular 
>> microarchitectures
They will ensure the machine they are running on is designed and 
certified to zero-tail-fill.

> Part of the reason for allowing nondeterminism even for the same 
> inputs to allow threads to migrate between cores that implement this 
> property differently.
valid, but not relevant for the original post.
>
> So, determining that you're currently running on a core that zeros the 
> tail isn't a guarantee that you can rely on tail zeroing for the 
> duration of the process' execution.
True. Ones need more extensive guarantees, thus a mode for zero 
tail-fill is a reasonable extension.
The non-determinism raises  a concern that a nefarious implementation 
could use the tail data as a side-channel vector. [duality intended].

There are many ways it can do so.

Stipulating that once the 1s are use, the rest of the tail must be 1s 
filled, will reduce the possibility or degree of side-channel exposure.

knightsifive, 2021-03-09:
While nobody has managed to change my opinion (that we should replace "1s" by "0s" in Sec. 3.3.3), I am convinced that this is ultimately a subjective matter, where my opinion differs from the consensus. That's fine.

I look forward to revisiting "tail zeroing" in RVV 2.0.

Thanks for the discussion.

closed 2021-03-09 by knightsifive