riscvarchive/riscv-v-spec #656: Generate illegal VLMAX with fractional LMUL

lihanglin, 2021-03-23:
The spec. says

> Implementations must support fractional LMUL settings for LMUL ≥ SEWLMUL1MIN/SEWLMUL1MAX, where SEWLMUL1MIN is the narrowest supported SEW value at LMUL=1 and SEWLMUL1MAX is the widest supported SEW value at LMUL=1. An attempt to set an unsupported SEW and LMUL configuration sets the vill bit in vtype.

Suppose there is an implementation: VLEN=128, ELEN=64
The instruction

```asm
  vsetvli a0, x0, e64, mf4
```

would result in VLMAX=1/2 (VLEN/SEW * LMUL).
The VLMAX is illegal and should assert vtype.vill.

Correct ?

Moreover, there is another implementation: VLEN=256, ELEN=64
The same instruction would have VLMAX=1 and is a legal form.

Same instruction with two different results and program should expect the behavior.

Correct ?

aswaterman, 2021-03-23:
The next sentence in the spec is relevant here: "For a given supported fractional LMUL setting, implementations must support SEW settings between SEWLMUL1MIN and LMUL * SEWLMUL1MAX, inclusive."  Here, we have LMUL * SEWLMUL1MAX = 16, so SEW=64 is not required to be supported.

Programmers don't need to know VLEN to write portable code: they should just avoid violating the above constraint, and only use LMUL >= 1 when SEW = SEW_LMUL1MAX.

lihanglin, 2021-03-24:
Clear. Thanks.

closed 2021-03-24 by lihanglin