riscvarchive/riscv-v-spec #658: Regarding VLMAX

GAYATHRIGP-1996, 2021-03-30:
How to set this maximum vector length? How come the programmer knows that 512(supposed to be) is the maximum vector length that an architecture will support?

GAYATHRIGP-1996, 2021-03-31:
Based on the implementation, maximum vector length support has to be specified first right?
For example maximum VLEN = 512bits  and the supported configurations are 8 64bit elements or 16 32 bit elements have to be specified first.

My doubt is how come the programmer knows the maximum number of bits supported in that design is 512 bits. There is a "misa" CSR in base RISC V privileged spec, like Is there any bit field to indicating the maximum VLEN supported in the design to the programmer?

kasanovic, 2021-04-08:
question was answered, so closing this issue.

closed 2021-04-08 by kasanovic