riscvarchive/riscv-v-spec #659: question about vasub

wzy971210, 2021-04-06:
Hi, I have a question about vasub. It is said that vasub can't be overflow.
> For vaaddu, vaadd, and vasub, there can be no overflow in the result. For vasubu, overflow is ignored and the result wraps around.

However, when sew=16, vxrm = rnu ,vs2= 0x7fff, vs1 = 0x8000, the vasub opeartion can be like this:

`
roundoff_signed(0x7fff-0x8000,1) 
= signed(0x7fff-0x8000) >> 1 + 1
= 0xffff>>1+1
=0x8000
`

The result is 0x8000, a negative value. Isn't it weird? A positive number minus a negative number equals a positive number.
 From my perspective, the result is overflow beacause  it is a value greater than 0x7fff, right?

aswaterman, 2021-04-06:
Because the LHS of the >> operator is signed, it's an arithmetic shift.  So the expression evaluates not to 0x8000 but to 0, which is what you'd hope (since averaging the largest-magnitude positive and negative numbers should result in approximately 0).

wzy971210, 2021-04-06:
Thanks for replying.
So you mean the addition/subtraction is **an unsigned operation** for _vasub_ and _vaadd_? 
The vasub opeartion can be like this:

`roundoff_signed(0x7fff-0x8000,1) =  (unsigned(0x7fff) - unsigned(0x8000)) >> 1 + 1 = 0xffffffff>>1+1 =0x0`

Is that right?

aswaterman, 2021-04-06:
No, I'm saying that the >> is an arithmetic shift.  So, going back to your earlier message, `(signed(0xffff) >> 1) + 1` becomes `0xffff + 1`, i.e., 0.

wzy971210, 2021-04-06:
I know it is an arithmetic shift. But what we here talk about is the signed or unsigned addition/subtraction.
I was thinking that for vasub, 
`0x7fff - 0x8000 = signed(0x7fff) - signed(0x8000) = 0x00007fff - 0xffff8000 = 0x0000ffff; (signed(0x0000ffff) >> 1) + 1 = 0x8000`
or
`0x7fff - 0x8000 = unsigned(0x7fff) - unsigned(0x8000) = 0x00007fff - 0x00008000 = 0xffffffff; (signed(0xffffffff) >> 1) + 1 = 0x0`

kasanovic, 2021-04-08:
There are only 16b in your example, so signed(0x0000ffff) is not possible,  should be signed (0xffff).

As spec says, difference between  vaadd/vasub and vaaddu/vasubu  is that former are treated as signed values and latter as unsigned values.  Required shift behavior follows.

wzy971210, 2021-05-13:
Hi, when I try to figure out this question on Spike, to see the result of this instruction , I see the result `0x8000`, not `0x0000`. So, please rethink and confirm the result about vasub?
[image]

aswaterman, 2021-05-13:
Spike is doing the right thing, and the description of the instruction needs a tweak.

The discrepancy you're seeing is because of rounding.  (If you didn't explicitly set the rounding mode, Spike defaults to RNU.)  We are computing `(32767 - -32768) / 2` == 32767.5.  When we round the result, we get 32767 for RDN or ROD, but we get 32768 for RNU or RNE.  Of course, 32768 cannot be represented in 16-bit two's complement.  When truncated to 16 bits, it looks like -32768.

So, the claim that `vasub` can't overflow needs qualification--it's true for RDN or ROD, but it's false for RNU or RNE.  I'll make a PR to fix the description.

wzy971210, 2021-05-13:
thanks for replying. The keypoint is the operands of  vasub and vaadd are both signed value. So, the operation for them, which is addition and shifting in this case, is signed-operation. We need to signed-extend them first.

commit be582adb (2021-05-13): Fix incorrect statement about vasub overflow

It's true that vasub can never overflow when rounding down or jamming,
but when rounding to nearest, overflow _can_ occur.

Resolves #659
commit e185ee31 (2021-05-13): Fix incorrect statement about vasub overflow

It's true that vasub can never overflow when rounding down or jamming,
but when rounding to nearest, overflow _can_ occur.

Resolves #659
closed 2021-05-13 by kasanovic