riscvarchive/riscv-v-spec #662: Widening and narrowing in load/store legality unspecified

brucehoult, 2021-04-11:
I can't find any words saying what happens if the element width in memory does not match the element width set by VSETVL{I}.

The introduction of "5. Vector Instruction Formats" states: "Vector loads and stores move bit patterns between vector register elements and memory. Vector arithmetic instructions operate on values held in vector register elements."

The words "bit patterns" suggest that no interpretation of the bits is made, and thus widening (at least!) on load or store is not possible. There is no way to know whether values are integers or floating point or whether an integer should be zero or sign extended. Moving a 32 bit FP value to a 64 bit element or vice versa is just nonsensical.

This suggests that the statically encoded element width in the load or store must match the dynamically set VTYPE.

Table 11 in "7.3. Vector Load/Store Width Encoding" shows "Mem Bits" always matching "Data Reg bits" for "Vector {8..1024}b element" loads and stores, so this supports the view that they must always match.

However, "7.1. Vector Load/Store Instruction Encoding" states: "Vector memory unit-stride and constant-stride operations directly encode EEW of the data to be transferred statically in the instruction to reduce the number of vtype changes when accessing memory in a mixed-width routine."

I don't understand how this can be true, unless the element width set in VTYPE and the element width in the load/store can be different.

Which beings us back to there apparently being no words describing what happens if they differ.

I'm unable to find examples in the various code samples where the elements width in load/store does not match that in the preceding VSETVL.

In particular, they do match in "6.4. Example of stripmining and changes to SEW"

There is one interpretation that allows all the above statements to be simultaneously true:

- if the memory element width is greater than the register element width then each memory element fills (or is filled from for stores) several consecutive elements in the register.

- if the memory element width is smaller than the register element width then each register element fills (or is filled from for loads) several consecutive elements in memory.

Under this interpretation the number of bits transferred to/from memory is the same as the current vector length, but the number of vector elements can be different. The memory element width would then be irrelevant for unit stride loads and stores (and a waste of encoding space), but will produce different results for strided or indexed loads and stores.

If this is the desired interpretation then it really needs some words to describe it.

If the desired interpretation is that the memory element width must always match the register element width then the above quoted statement in 7.1 needs to be removed.

If the quoted statement in 7.1 is intended to stand then some words about how this works and one ore more examples are needed -- and Table 11 needs changing.

aswaterman, 2021-04-12:
Suppose the current SEW is A, current LMUL is B, and you execute a vector memory access with statically encoded EEW of C.  Doing so is exactly equivalent to changing SEW to C, changing LMUL to B*C/A, then executing that same memory access.  (And then resetting SEW and LMUL back to A and B.)

brucehoult, 2021-04-12:
OK, so VL stays the same, and the amount of data transferred changes in the ratio C/A. So it's useful for storing the results of a widening or narrowing operation without changing VTYPE.

This definitely needs specifying (because I believe it currently is not), and an example.

brucehoult, 2021-04-12:
ok, so "The corresponding EMUL is calculated as EMUL = (EEW/SEW)*LMUL." does seem to cover this, but it's not really obvious.

Could do with words added to "5.2. Vector Operands" stating that loads and stores are affected as well as widening and narrowing arithmetic.

aswaterman, 2021-04-12:
Yeah, the goal was just to cut down on vtype changes.  I agree that it's currently specified sufficiently, so there's no urgent need to update the spec, but once it's ratified, more explanatory material might be appropriate.

There is an example program that depends on this facility (https://github.com/riscv/riscv-v-spec/blob/95e44cb1688ed59c18d5ddc15915aed4ef8df4ac/vector-examples.adoc#example-with-mixed-width-mask-and-compute) but without commentary it's also not entirely obvious.

brucehoult, 2021-04-12:
I don't agree this example depends on this facility. If it does then it's completely non-obvious to me.

The vle8 is preceded by a matching width vsetvli e8,m1. The vmslt of course produces 1 bit results but that is a different issue.

The vle32 and vse32 are preceded by a matching width vsetvli e32,m4.

So there is no vtype sew to memory element width mismatch here.

I also don't agree that the spec should be made more clear only after ratification. If you hope to get meaningful feedback from "public" review (i.e. people not on the Working Group) then the spec needs to be as clear as possible. Not to mention to ensure people working on implementations get it right.

aswaterman, 2021-04-12:
Oops, you’re right.

I’ll leave it to others to debate the matter of augmenting the spec with non-normative text explaining the use case.

closed 2021-04-18 by brucehoult