riscvarchive/riscv-v-spec #668: the cycles of vector instructions?

ALLap97, 2021-04-25:
Hi! I want to know some information about the execution cycles of the vector instructions.
Does it take more cycles to execute a vector instruction than a scalar instruction?
And does the difference of lmul affect the number of execution cycles of the same vector instruction?
Is there any specific data introduction to the number of execution cycles of different vector instructions？(like Arm NEON intructions)

brucehoult, 2021-04-25:
That all depends on the design of an individual CPU. The RISC-V Vector extension is intended to be appropriate for and allow the same program to run on a very wide range of implementation points.

Some implementations might process one vector element per clock cycle -- some might take many clock cycles to process each vector element. Some will process 64 bits or 128 bits or 256 bits or 1024 bits or 4096 bits of data in one clock cycle. Some implementations might process 2 or 3 or 4 vector instructions in each clock cycle.

ALLap97, 2021-04-26:
but in the same CPU, does it take more cycles to execute a vector instruction than a scalar instruction? And does the difference of lmul affect the number of execution cycles of the same vector instruction in the same CPU?
Thanks!

brucehoult, 2021-04-26:
It depends on the design of the CPU.

If the vector registers are built using the same technology as the scalar registers, and the vector unit has an ALU for every data item in the vector, then it is entirely possible for a vector instruction to take the same amount of time as the corresponding scalar instruction.

Some other machine might only have enough ALUs to process 1/4 of a vector at one time, and use 4 clock cycles to process the entire vector register (and 8 or 16 or 32 clock cycles with longer LMUL). But they might have longer vector registers than the machine I described in the paragraph above.

Some designers might observe that most vector code doesn't need 32 variables and so the common case will be for an LMUL of 2 or 4, and they might design around that, providing shorter vector registers but enough ALUs to process all the elements from 2 or 4 vector registers in parallel.

That's the beauty of RVV. The CPU designer has huge scope to do things different ways, depending on the market and algorithms they are targeting.

brucehoult, 2021-04-26:
As a concrete example, the only mass-produced RISC-V Vector CPU I have used so far is the Allwinner D1. which uses the Alibaba T-Head C906 RISC-V core, a single-issue in-order core with a classic 5 stage pipeline and VLEN=128, running at 1.0 GHz. This is a low end low cost chip, little more than a microcontroller (but with an MMU and vector unit and DRAM support) that will be sold on board with prices as low as $10 or $12 or maybe even less.

From the C906 manual, simple arithmetic operations take 3 clock cycles at LMUL=1. The manual doesn't actually specify the timings for longer LMUL (that I can find). These simple operations include VAND, VOR, VXOR, VMERGE, VMV, VMAND, VMNAND, VMANDNOT, VMXOR, VMOR, VMOR, VMORNOT, VMXNOR, VMPOPC, VMFIRST, VMSBF, VMSIF, VMSOF, VIOTA, VID, VREDAND, VREDOR, VREDXOR, VSLL, VSRL, VSRA, VADD, VSUB, VRSUB, VMSEQ, VMSNE, VMSLT (etc), VMIN{U}, VMAX{U}, VEXT, VSLIDEUP, VSLIDEDOWN, VRGATHER, VCOMPRESS, VFADD, VFSUB (etc), FMAs in .VF flavour. You get the idea.

Things that take 4 cycles include: VRED{SUM,MIN{U},MAX{U}},  FMAs in .VV flavour, 

Unit stride loads take minimum 2 cycles (I think more if cache lines are crossed or something). Unit stride stores take minimum 1 cycle. Strided loads take >= 2 + VLEN/SEW. Strided stores take >= 1 + VLEN/SEW. Unit stride segmented loads take >= 2 + nf. Unit strinde segmented stores take >= 1 + nf.

Integer multiplies and multiply-accumulates take 3 for SEV <= 16, 4 for 32 bit.

Integer div/rem take 3-21. FP DIV/RDIV/SQRT take 4-17. FP unordered reductions take 3*log2(VLEN/SEW),. Ordered reductions take 3*(VLEN/SEW).

The following loop executes at 16.25 clock cycles per iteration (64 bytes). Perhaps I didn't have optimal alignment in my experiments.

0000000000000000 <memcpy>:
   0:	86aa                	mv	a3,a0

0000000000000002 <.L1^B1>:
   2:	00267757          	vsetvli	a4,a2,e8,m4,d1
   6:	12058007          	vlb.v	v0,(a1)
   a:	95ba                	add	a1,a1,a4
   c:	8e19                	sub	a2,a2,a4
   e:	02068027          	vsb.v	v0,(a3)
  12:	96ba                	add	a3,a3,a4
  14:	f67d                	bnez	a2,2 <.L1^B1>
  16:	8082                	ret

The following loop executes at 43.75 clock cycles per iteration (64 bytes). Same comment on alignment.

000000000000001a <strcpy>:
  1a:	862a                	mv	a2,a0

000000000000001c <.L1^B2>:
  1c:	00207057          	vsetvli	zero,zero,e8,m4,d1
  20:	03058207          	vlbuff.v	v4,(a1)
  24:	c2002373          	csrr	t1,vl
  28:	62403057          	vmseq.vi	v0,v4,0
  2c:	560026d7          	vmfirst.m	a3,v0
  30:	959a                	add	a1,a1,t1
  32:	5a01a057          	vmsif.m	v0,v0
  36:	00060227          	vsb.v	v4,(a2),v0.t
  3a:	961a                	add	a2,a2,t1
  3c:	fe06c0e3          	bltz	a3,1c <.L1^B2>
  40:	8082                	ret

Again, this is a low end chip. Something designed for phones / tablets / PCs / servers / supercomputers will have better performance than this.

ALLap97, 2021-04-27:
OK! 
Thanks a lot!

closed 2021-04-27 by ALLap97