riscvarchive/riscv-v-spec #669: Note vl=0 behaviour in section 5.4

rofirrim, 2021-04-29:
Section 5.4 gives the following definitions

```
    for element index x
    prestart(x) = (0 <= x < vstart)
    body(x)     = (vstart <= x < vl)
    tail(x)     = (vl <= x < max(VLMAX,VLEN/SEW))
    mask(x)     = unmasked || v0.mask[x] == 1
    active(x)   = body(x) && mask(x)
    inactive(x) = body(x) && !mask(x)
```

however those might be mildly in contradiction with section 3.4 that says

> Note: As a consequence, when `vl=0`, no elements are updated in the destination vector register group, regardless of `vstart` 

While a valid reading is that 3.4 overrules 5.4 and so the text is fine, I think it is too easy to miss that fact and assume that under `vl=0` a common vector operation will be "all tail" along with empty "prestart" and "body". 

We may not need to change the definitions but I'd suggest to repeat the note or add a new one reminding the reader that `vl=0` is kind of special.

kasanovic, 2021-06-09:
I don't see a contradiction between 5.4 and 3.4. Can you point out the conflicting case?

kasanovic, 2021-06-09:
Please review https://github.com/riscv/riscv-v-spec/commit/473f8e5b4ef9531d6fc2cafd1b2cb76d6c4fa5fc#diff-ea57dd7a8daf0aa62f553688c1970c8e6608945d25597f8661c5ea6670fb509c to see if this has adequately clarified these definitions.

rofirrim, 2021-06-09:
Hi @kasanovic, yes with that change I believe it should be harder to miss. Thanks!

closed 2021-06-09 by rofirrim