riscvarchive/riscv-v-spec #676: inconsistency: mask-agnostic or mask-undisturbed?

vbx-glemieux, 2021-05-14:
Section 3.3.3 has several notes ... two of the notes have conflicting information. One suggests mask operations are always tail-agonostic, the other suggests both tail-agnostic and tail-undisturbed are possible. This needs to be reconciled.

David-Horner, 2021-05-14:
First paragraph in section 3.3.3:
> These two bits modify the behavior of destination tail elements and destination **inactive masked-off** elements respectively
> during the execution of vector instructions. The tail and inactive sets contain element positions that are not receiving new
> results during a vector operation, as defined in Section Prestart, Active, Inactive, Body, and Tail Element Definitions

appears to set up the definitions sufficiently especially given the reference to section 5.4 hyperlinked using its title,

I believe adding  5.4 to the Section identifier would assist those with hardcopy like media [braille, etc.] .

However, with that setup I cannot identify the actual passages that are tripping you up.
Please identify. pasting as quotation should work adequately [I hope, as that is what I did].

ubc-guy, 2021-05-17:
Not sure of the best solution, but the problem is that "inactive" is used as a synonym for "masked-off".

The name of the field (vma), and the settings (ma, mu) all refer to "masked off", and these are frequently used in section 3.3.3 and elsewhere regarding the agnostic/undisturbed polity. Yet, Section 5.4 tries to define a new term, Inactive, which also makes a lot of sense but is seldom used. The table in section 3.3.3 using the column heading "Inactive Elements", yet the body always refers to "mask-agnostic" or "mask-undisturbed" as the settings.

To avoid confusion, it is typical to pick a single term and stick with it. I think either term is are fine, and you can even use one term to define the other, but using them interchangeably is hazardous.

ubc-guy, 2021-05-17:
I've proposed a distinction between masked-off and inactive elements, and submitted it as a pull request.

briefly:

new definitions:
inactive elements are those elements not updated in an instruction invocation (prestart + masked-off + tail)
active elements are those elements updated (body && selected by a mask)

the inactive element set is now larger, as it includes both tail and prestart elements. the rest of spec was updated to be more precise about the term 'inactive'. clarify that prestart elements are always undisturbed.