riscvarchive/riscv-v-spec #677: bitwise logical ops: IMM vs UIMM, reserve useless IMM values

vbx-glemieux, 2021-05-15:
as a programmer, i expect UIMM (not IMM) for all bitwise ops. this is because they are bitwise, ie operands are to be interpreted as bags of bits, two's complement integers.

the spec says otherwise, that they are all IMM:

# Bitwise logical operations.
vand.vi vd, vs2, imm, vm   # vector-immediate
vor.vi vd, vs2, imm, vm    # vector-immediate
vxor.vi vd, vs2, imm, vm    # vector-immediate

a note in the spec gives  explains that vxor can be used to do complete inversion using IMM=-1. this seems to be the sole reason for using IMM everywhere.

however, for the vand/vor instructions, it's not so clear why it should be treated as signed, other than to treat them all the same way. (I can't think of any other strong rationalization for IMM vs UIMM for these instructions, as all possible interpretations seem to have strengths and weaknesses.)

since this is counter-intuitive, I believe an explicit note/rationale should be added to clarify this is intentional and not a typo or oversight.

----

my preferred solution is actually to change all of these to UIMM to match programmer intuition.

to enable vnot, define it as the encoding of vxor with UIMM=0. otherwise, this is a logical operation that would never be used.

finally, define the encoding of vor with UIMM=0 to be reserved, making it available for a single-operand integer instruction in the future (eg: popcount, find-first-one, count-leading-zeros, square root, log2, etc).

aswaterman, 2021-05-15:
Sign-extending the immediates is also the more useful choice for `vand.vi`, as it is frequently used to mask off individual bits (i.e., most of the bits in the mask should be set).

FWIW, it is consistent with the base ISA, for which the andi/ori/xori immediates are also sign-extended.  In studying this years ago, I found that making the immediates signed was beneficial for code size overall (especially for `xori` and `andi`; for `ori` it was about the same either way).

ubc-guy, 2021-05-17:
if it is already consistent with base ISA, I would not change this now -- I should have checked this.

however, vor and vxor are NOPs when IMM=0. these opcodes should be reserved.

vand with IMM=0 is also pretty useless (but it's not a NOP). I'd suggest reserving it as well.

kasanovic, 2021-06-09:
Separate issues cover NOPs/Hints. Closing this one.

closed 2021-06-09 by kasanovic