riscvarchive/riscv-v-spec #678: slt ops: unsigned comparison should use UIMM instead of IMM

vbx-glemieux, 2021-05-16:
unsigned comparison operators with immediate operands seem to use IMM instead of UIMM:

vmsleu.vi vd, vs2, imm, vm   # Vector-immediate
vmsgtu.vi vd, vs2, imm, vm   # Vector-immediate

is this intentional, or should it be UIMM?

aswaterman, 2021-05-16:
As with SLTIU in the base ISA, the immediates are sign-extended even though they represent unsigned quantities.

ubc-guy, 2021-05-17:
now a part of me wants to change the base ISA as well :P

as much as this displeases me, I can understand the rationale. to avoid perception of a typo, should a note be added stating this is done for consistency with base ISA?

aswaterman, 2021-05-17:
The NOTE immediately below the assembly does make it clear, I think.

ubc-guy, 2021-05-17:
oops that note stresses my patience due to it's length and intricate content, so I didn't read it carefully. yes, I see now that it mentions "always sign-extended", but doesn't say WHY. this is also a separate concept from the beginning of the note which talks about the lack of a .vi form for vmslt{u}.

there should only be one concept per note, splitting this one up into 2 or more:  (a) omission of vmslt{u}.vi and range limits, (b) 5b sign-extended immediates with implication on values <0, and (c) all of the useless opcode encodings, eg when IMM=0 or 2^SEW (and I recommend that these useless encodings be reserved for future instructions)

kasanovic, 2021-06-04:
@ubc-guy Please submit pull request with better text

commit 4098674a (2021-07-15): Improved description of unsigned slt instructions.
Closes #678.
closed 2021-07-15 by kasanovic