riscvarchive/riscv-v-spec #681: vfredsum and vfredosum inconsistency

ubc-guy, 2021-05-16:
Except for vfredsum, all instructions that have both ordered and unordered versions use a 'u' and an 'o' to distinguish the two variants.

vforedsum is the ordered version, which is IEEE-compliant.

vfredsum should be renamed vfuredsum

I believe the default behaviour of an "unspecified order" reduction (ie, vfredsum) should be the "safe" IEEE-compliant one (vforedsum), but that's not currently the case.

aswaterman, 2021-05-16:
Valid point.  The ordered/unordered loads and stores were defined after most of the rest of the spec was written, which introduced this inconsistency.

I'm not opposed to renaming vfredsum to vfredusum.  But to avoid changing existing code, vfredsum would need to continue being accepted as an alias for vfredusum.

Given closeness to 1.0, I'm also not opposed to doing nothing :P

ubc-guy, 2021-05-17:
I would suggest the official name be changed, since it clarifies things for consistency without introducing a functional change.

I would also suggest adding a note that vfredsum is an alias for vfuredsum, adding this alias is for pre-1.0 compatibility and may be deprecated in the future.

commit a66753c6 (2021-06-09): Changed assembler mnemonic of unordered floating-point reductions `vfredsum` and `vfwredsum` to `vfredusum` and `vfwredusum` respectively.  Older names kept as aliases.
Closes #681.
closed 2021-06-09 by kasanovic