riscvarchive/riscv-v-spec #682: fixed-point reductions and dot products

ubc-guy, 2021-05-16:
Is there enough opcode space to support:

a) fixed-point reductions, possibly for both averaging and saturating versions (very useful for ML workloads)

b) dot product for both integers and fixed-point, eg:
vdotprod vd, vs2, vs1, vm # vd[0] = sum( vs2[i] * vs1[i] ) 

i know the dot product has been discussed before, but i'm disappointed by it's omission. the current reduction instructions do N+1 ops, with the +1 being the scalar seed. a dot product can do 2N ops, using already existing multipliers and the reduction tree.

if you wish to keep the ability to "build on" a past reduction, then can combine it with a slide1up operation to build an array of values which can later be reduced:

vdotprodslide1up vd, vs1, vm # vd[0] = sum( vs2[i] * vs1[i] ); vd[i+1] = vd[i]

aswaterman, 2021-05-17:
I feel like (a) is reasonably well catered to already, with e.g. `vwredsum` + `vnclip`.

I suspect vdot will reappear in a future extension once the EDIV stuff is revived.

ubc-guy, 2021-05-17:
(a) you're right, thanks, I should have thought harder.

(b) understood.

closing this...

closed 2021-05-17 by ubc-guy