riscvarchive/riscv-v-spec #686: The masked versions (vm=0) of mask logical instructions are not reserved?

zhongqibiao, 2021-05-21:
In section 16.1,  it says "Vector mask logical instructions are always unmasked so there are no inactive elements". But unlike the other unmasked only instructions, the spec does not state the masked versions of these mask logical instruction are reserved. Should illegal instruction exception be raised when executing the masked versions of mask logical instructions?

I have run a test on spike,  vmand.mm with vm=0 will not cause illegal instruction exception.

Spike output text:
core   0: 0x00000000000101e2 (0x66942557) vmand.mm v10, v9, v8
: 
core   0: 0x00000000000101e6 (0x64942557) vmand.mm v10, v9, v8
:

aswaterman, 2021-05-21:
I'll make a PR that clarifies this in the spec.

Spike's behavior is technically not incorrect, since trapping reserved encodings is not required.  But we should still change its behavior at some point.

zhongqibiao, 2021-05-22:
Thanks for your reply.
I was wrong about reserved encoding must cause illegal instruction exception. It's just a recommendation.

Agree that it's better if spike keep the same behavior when executing  all reserved encodings.

PR #687 (2021-06-09): vm=0 encodings should be reserved for mask-logical instructions
I guess this is technically a semantic change, not a clarification, but I believe it is what @kasanovic intended in writing "vector mask logical instructions are always unmasked".

Resolves #686
commit a38383f2 (2021-06-09): vm=0 encodings are reserved for mask-logical instructions

Resolves #686
closed 2021-06-09 by kasanovic