riscvarchive/riscv-v-spec #697: zve64* and zve32* complicated the arch string parser logic

kito-cheng, 2021-06-18:
Currently we don't have any extension name contain digit, zve64* and zv32* are the first extension named with digit, that cause the current arch string parser logic, since we assume the any digit are version before*, parser logic will become complicate without this assumption.

* LLVM and GCC parser are using similar logic to parse arch string, go to parsing version mode once got digit.

nick-knight, 2021-06-18:
The ISA Manual (I:25.6) says:
> Standard extensions can also be named using a single “Z” followed by an alphabetical name and an optional version number.

My reading of this implies that extension names like `Zve32` are illegal.

David-Horner, 2021-06-19:
Easily solved with zvesixtyfour?

kito-cheng, 2021-06-19:
@nick-knight Hmmm, yeah, ISA spec seems not forbid extension name contain with digit, but with this constraint parser would be easier to implement.

@David-Horner `zvesixtyfour` good to me, there is no digit in the name :P but I know it's really verbose than zv64*...

aswaterman, 2021-06-21:
@kito-cheng It's implicitly forbidden by the current spec, because the current spec says the number will parse as a version number (i.e. Zve64 will be unambiguously interpreted as Zve, version 64).

kito-cheng, 2021-06-29:
Some offline chat with @aswaterman :
Kito: I know `zve64` is pretty ambiguous with version, so how do you think about `zve64x`?
Andrew: I think Krste needs to change the ISA strings to avoid numbers, changing parser doesn't makes sense to me, because it violates the spec
...
Kito: but I admit `zve64*`/`zve32*` is really expressive
Andrew: It's unfortunate that it's both expressive and invalid, maybe zved and zvew (doubleword / word)

aswaterman, 2021-06-29:
PS.  Given that we require `_` to separate multi-letter extensions, I agree that something like `zve64x` would parse unambiguously, because of the `x`.  `zve64x2p0` would give a version number in this case.

DSHorner, 2021-06-29:
>
> PS. Given that we require |_| to separate multi-letter extensions, I 
> agree that something like |zve64x| would parse unambiguously, because 
> of the |x|.

> |zve64x2p0| would give a version number in this case.
Is the version number 2p0 in this case? asking for clarification?

there is roman numerals: XXXII and LXIV .

Greek - like roman numerals but more characters if modern form used.

There are other numeral representation.

三十 二 and 六十 四 for 32 and 64 respectively in Mandarin/Cantonese.

. סד and ל ב ‎Hebrew for 32 and 64 respectively

The strings are assumed to be unicode, right?

I prefer the chinese in that it is: very expressive and intuitive; 
reasonably concise; and acknowledges  major efforts from Asian participants.

I am open to other Asian representations, but the alternate dialects are 
many and represent fewer portions of the populations there.

kito-cheng, 2021-08-19:
Created a PR for clarify the rule.
https://github.com/riscv/riscv-isa-manual/pull/718

closed 2021-08-19 by kito-cheng