riscvarchive/riscv-v-spec #698: about designation on indexed ordered loads/stores vs zero strided

jnk0le, 2021-06-21:
In description of strided instructions:

>When rs2=x0, then an implementation is allowed, but not required, to perform fewer memory operations than the number of active elements, and may perform different numbers of memory operations across different dynamic executions of the same static instruction.
>When rs2!=x0 and the value of x[rs2]=0, the implementation must perform one memory access for each active element (but these accesses will not be ordered).

Except from splat-from-memory (which can be done bu unordered indexed load or load to scalar + splat), the rest cases are useless from software perspective. 
What is the purpose of requiring to perform all acceses in case of `rs2!=x0&&x[rs2]=0`? (this is how it will be typically implemented, but why normative text)

>Note: To provide ordered vector accesses to a strongly ordered IO region, the ordered indexed instructions should be used. 

>Note: When repeating ordered vector accesses to the same memory address are required, then an ordered indexed operation can be used.

It implies that all indexed ordered loads/stores **must** be performed element by element, suppressing any possible optimizations that honour ordering rules. (Though, there is nothing about it inside indexed chapter)

e.g. 
Initial purpose of indexed ordered stores was to have only the "oldest" element written at given index. (I think that covers 90+% of use cases for ordered store)
- In case of repeating index the hardware can just write the oldest element from sequence (without crossing element with different index due to being ordered).
- Unit stride increasing indexes can be merged into a single bigger transfer.
- spatial permutation network in lsu. etc.

With above normative notes those optimizations are forbidden.

I think that the `rs2=x0` cases of strided load/stores could be the ones that performs all acceses in order, one by one and leave `rs2!=x0&&x[rs2]=0` up to implementation choice, as that's harder to decode.

If not then indexed load/store chapter needs at least more normative text.

nick-knight, 2021-06-21:
> What is the purpose of requiring to perform all acceses in case of rs2!=x0&&x[rs2]=0?

I was not closely involved in this part of the architecture, but I think I can dream up a use-case. Suppose we're constructing a histogram of data stored in an external device, accessed by a memory-mapped register. We want to perform all accesses, but the order doesn't matter.

I'm afraid I can't address your other comments.

jnk0le, 2021-06-24:
Another idea: add weakly ordered indexed stores (have only the "oldest" element written at given index) and keep the current ordered instractions according to the normative notes.

kasanovic, 2021-06-25:
rs2=x0 can be detected earlier in decode and used to create a specialized "splat" uop in microarchitecture.   Hardware can simply treat rs2!=x0, rs2=0 case the same as for any other value in rs2 without needing to check rs2 value.

Microarchitectures can still perform optimizations in all these cases to reduce number of microarchitectural memory reads as long as result is same as some architecturally legal result.   The language is written to intersect with the MCM.  Concerns are mainly about visibility from other threads or devices., and in preserving semantics of original code when autovectorizing.

jnk0le, 2021-06-27:
>Microarchitectures can still perform optimizations in all these cases to reduce number of microarchitectural memory reads 
`rs2!=x0&&x[rs2]=0` case, is required to perform all acceses.

aswaterman, 2021-06-27:
> `rs2!=x0&&x[rs2]=0` case, is required to perform all acceses.

That's not entirely true.  As usual, the microarchitecture can do whatever it wants, as long as it's indistinguishable from some architecturally valid execution.  For idempotent memory regions, coalescing a sequence of loads to the same address into a single load is architecturally indistinguishable from stopping the other harts then performing the sequence of loads one-by-one.  So that's a valid optimization.  It is only for non-idempotent memory regions that the microarchitecture really must perform all accesses.

jnk0le, 2021-06-29:
The spec says:
>When rs2!=x0 and the value of x[rs2]=0, the implementation must perform one memory access for each active element (but these accesses will not be ordered).
> Note: Compilers must be aware to not use the x0 form for rs2 when the immediate stride is 0 if the intent to is to require all memory accesses are performed.

My interpretation is that it "must" perform all acceses.

The other thing is how the (generic/licensed) core is going to distinguish the types of memory regions, hanging on the generic busses.

aswaterman, 2021-06-29:
The basic principle of microarchitecture is that the implementation can do whatever it wants to do as long as it's indistinguishable from an architecturally valid execution.  The spec describes only the architectural requirements.

kasanovic, 2021-07-10:
@jnk0le distinguishing idempotent versus non-idempotent is well-understood issue in IP core integration (as well as other attributes such as coalescing writes etc.).  For non-idempotent regions, the "must perform all accesses" is important as accesses may have side-effects beyond load and store.   For idempotent regions (e.g., regular main memory), any implementation that appears correct is allowed, i.e., not possible to observe any deviation from spec using other cores.  Closing issue.

closed 2021-07-10 by kasanovic